ev3dev/ev3dev #1349: ev3dev : problem with USB connection under Debian10

forhang, 2019-11-17:
Hello,

**CONTEXT :**
I'm facing a problem since several days. Last year, I had jessie's ev3dev version and I was under mint 17, everything was fine.

However, I passed to debian10 on August and also burned buster's ev3dev version (kernel 4.14.96-ev3dev-2.3.2-ev3 according to the brickman). Unfortunately, I'm now unable to connect through USB with a ssh connection.
Finally, I made again a new installation of debian10 from scratch without personal setting and made the following test.

**QUESTION :**
How can I setup a basic ssh connection between debian10 and the ev3dev's brick?
Will it be an automatic connection (ie if I unplugged and plug again the EV3 brick, should I make a setup again?).
I really appreciate any help.

**WHAT I ALREADY DID :**
Of course I went to the getting-started's page (https://www.ev3dev.org/docs/getting-started/)  then I read (https://github.com/topikachu/ev3dev/wiki/Connecting-via-USB-to-Linux-host-with-NetworkManager) and (https://github.com/topikachu/ev3dev/wiki/Setting-Up-Linux-USB-Ethernet-Networking) and several questions with usb, ssh, connection keys words (as [ev3dev/ev3dev/issues/1191] or [ev3dev/ev3dev/issues/1264]).

With the tutorial of Topikachu to set-up usb ethernet networking (3rd link), I made these steps :
1) plugged the brick and select wired connection
2) it showed me 169.254.226.30 : status connected
3) nb : I didn't find the ev3dev.rc.local file on the SD card neither with find nor grep command? Is it mandatory, where is it? Should I burn again the SD card?
4) entered some commands

`@debian:~$ ls /sys/class/net`
_enp0s18f2u4  enp63s0  lo_

`@debian:~$ /sbin/ifconfig`
_enp0s18f2u4: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.1  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::1016:53ff:fe4e:d015  prefixlen 64  scopeid 0x20<link>
        ether 12:16:53:4e:d0:15  txqueuelen 1000  (Ethernet)
        RX packets 278  bytes 142722 (139.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 47  bytes 5440 (5.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
  [...]_

`@debian:~$ sudo dmesg`
_[...]
[25725.156838] usb 1-4: new high-speed USB device number 34 using ehci-pci
[25730.428826] usb 1-4: device descriptor read/64, error -110
[25746.044821] usb 1-4: device descriptor read/64, error -110
[25746.280825] usb 1-4: new high-speed USB device number 35 using ehci-pci
[25751.420814] usb 1-4: device descriptor read/64, error -110
[25767.036793] usb 1-4: device descriptor read/64, error -110
[25767.144869] usb usb1-port4: attempt power cycle
[25767.588793] usb 1-4: new high-speed USB device number 36 using ehci-pci
[25778.380682] usb 1-4: device not accepting address 36, error -110
[25778.508749] usb 1-4: new high-speed USB device number 37 using ehci-pci
[25789.132737] usb 1-4: device not accepting address 37, error -110
[25789.132781] usb usb1-port4: unable to enumerate USB device
[25789.516753] usb 4-1: new full-speed USB device number 11 using ohci-pci
[25804.976740] usb 4-1: device descriptor read/64, error -110
[25820.592641] usb 4-1: device descriptor read/64, error -110
[25820.884632] usb 4-1: new full-speed USB device number 12 using ohci-pci
[25836.464610] usb 4-1: device descriptor read/64, error -110
[25850.128702] usb 1-4: new high-speed USB device number 38 using ehci-pci
[25850.287053] usb 1-4: New USB device found, idVendor=0694, idProduct=0005, bcdDevice=30.01
[25850.287062] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[25850.287066] usb 1-4: Product: EV3+ev3dev
[25850.287070] usb 1-4: Manufacturer: LEGO Group
[25850.287073] usb 1-4: SerialNumber: 0016534ED015
[25850.294734] cdc_ether 1-4:1.0 usb0: register 'cdc_ether' at usb-0000:00:12.2-4, CDC Ethernet Device, 12:16:53:4e:d0:15
[25850.307541] cdc_ether 1-4:1.0 enp0s18f2u4: renamed from usb0_

`@debian:~$ sudo ifconfig enp0s18f2u4 192.168.2.1`

5) from here, when I enter
`@debian:~$ ssh robot@192.168.2.1`

The system asks me the password, but neither 'maker' nor 'r00tme' works (even if I try to login under root's account). I'm so rejected with a permission denied.

What's the problem?
I'm certainly missing an obvious setup but be sure it's not so obvious for me =). Thanks for any help.

**DESKTOP :**
 - OS and version: Debian buster 10 (`@debian:~$ uname -a` Linux debian 4.19.0-5-amd64 # 1 SMP Debian 4.19.37-5 (2019-06-19) x86_64 GNU/Linux with LXDE-PCManFM desktop)
 - Connection Type: ssh through USB
 - Software: openssh-client (version:7.9p1-10+deb10u1)

**ROBOT :**
 - Device: LEGO MINDSTORMS EV3
 - Versions: ev3dev-stretch-ev3-generic-2019-03-03 ; kernel 4.14.96-ev3dev-2.3.2-ev3

HomeIT4all, 2019-12-08:
This looks not right to me:

it showed me 169.254.226.30 : status connected

169.254.226.30 is an apipa address.

Can you change this address (Maybe to 192.168.2.2)?

Regards

forhang, 2020-01-14:
Hi, 

sorry for the late reply and thanks for the advise.

With your comment, I change the IP address in the field:
> All Network Connections
>>Wired
>>>IPv4

and choose Linux default setting (it now shows me 10.42.0.3 at the first line) and I'm able to connect.

By the way, I still need more testing on another similar computer to be sure it's ok.

B. Regards

forhang, 2020-07-04:
Hi,

since january, the problem was not fixed on my main computer. 
### To sum up:

Link communication seemed to be good, as after a manual link
>`sudo ifconfig enp0s20f0u2u4c2 10.42.0.3`

the interface name was linked to 10.42.0.3 and to the mac adress of the ev3Robot
>`sudo ifconfig -a`
enp0s20f0u2u4c2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.42.0.3  netmask 255.0.0.0  broadcast 10.255.255.255
        inet6 fe80::1016:53ff:fe4e:d015  prefixlen 64  scopeid 0x20<link>
        ether 12:16:53:4e:d0:15  txqueuelen 1000  (Ethernet)
        RX packets 19  bytes 1386 (1.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 56  bytes 6491 (6.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

and last the ping was ok
>$ `ping 10.42.0.3`
PING 10.42.0.3 (10.42.0.3) 56(84) bytes of data.
64 bytes from 10.42.0.3: icmp_seq=1 ttl=64 time=0.037 ms
64 bytes from 10.42.0.3: icmp_seq=2 ttl=64 time=0.039 ms
64 bytes from 10.42.0.3: icmp_seq=3 ttl=64 time=0.041 ms
^C
--- 10.42.0.3 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 30ms
rtt min/avg/max/mdev = 0.037/0.039/0.041/0.001 ms

but ssh connection did not recognize the password 'maker'
>`ssh robot@ev3dev.local`
The authenticity of host 'ev3dev.local (10.42.0.3)' can't be established.
ECDSA key fingerprint is SHA256:nAMelufyaUZjyhIzLZGpEvJOHlAbSrJ1MPlXfrTJFds.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'ev3dev.local' (ECDSA) to the list of known hosts.
robot@ev3dev.local's password: 
Permission denied, please try again.
robot@ev3dev.local's password: 

With my other computer which has Mint (Linux 4.15.0-76 generic # 86-Ubuntu SMP x86_64 GNU/Linux), I have not connectivity problem.

After other investigation and thanks to @dlech advise [on issue/942](https://github.com/ev3dev/ev3dev/issues/942#issuecomment-325157487), problem seems to be fixed.

### What it has been done:
- installation of `dnsmasq`, `network-manager` and `network-manager-config-connectivity-debian` packages
- tethering gadget on
- ipv4 set with default windows configuration (why it doesn't work with Linux's one? -> I don't know)

Let see the last on another computer at work office…