ev3dev/ev3dev #1350: Windows 10 USB device not recognized

BichonCby, 2019-11-17:
Hello,

I'm trying (for the second time) to use ev3dev. First time, months ago, I did connect with USB and it worked fine.
Now I'm trying again because I want to test python programming on ev3 so I did the same steps to connect on my PC running windows 10.
The problem is that the EV3 is not recognized with ev3dev microSD despite (it's OK with ev3 LEGO OS). 
I only see an unspecified device named "Remote NDIS Compatible Device #2" but I can't configure network settings because there is no menu proposed on right click (even if I wait long, windows doesn't know what this device is).

I tested my brick on another PC with windows 10 and it works, so my brick is OK and my ev3dev installation seems ok too.

how can I make it work on my PC? Do I have to delete any old Windows configuration?

HomeIT4all, 2019-11-18:
Try that. Say no and then okay.
[image]

BichonCby, 2019-11-18:
I don't have this menu as my device is not recognized. That's what I have :
[image]

HomeIT4all, 2019-11-18:
Okay see my Picture:
Delete your EV3 Driver. If you have 2 then delete both.
Change the other Setting to no and restart. After try again.
[image]

BichonCby, 2019-11-18:
Oh yes, thanks, it works. I deleted my ev3 driver, that seemed corrupted, and then I plugged ev3dev and it was recognized by windows. 

Danke sehr ;-)

closed 2019-11-18 by BichonCby