ev3dev/ev3dev #1351: ev3 gyro sensor returning 0-255 instead of signed 16bit degrees

salamanders, 2019-11-21:
I can work with this, but was wondering if it was normal.

BrickPi3, ev3dev, lejos, lego-ev3-gyro sensor, I’m getting odd readings. I was expecting degrees, but instead get 0-255 as I tilt the sensor.

From the [ev3-gyro-sensor docs](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#lego-ev3-gyro-sensor), it looks like it should be a signed 16-bit integer:

Which isn’t quite ideal, I can’t tell if it went in a circle, and the resolution isn’t as good as 360 degrees.

[My readings](https://docs.google.com/spreadsheets/d/1VTPyiQIeo71Cw6UfT2cIGR_BHopeKY8BzKksNEo2Ii8/edit?usp=sharing) over a few seconds,  I was rocking it back and forth.

I confirmed this was the same reading from `/sys/bus/lego/devices/spi0.1:S4:lego-ev3-gyro/lego-sensor/sensor0/device/lego-sensor/sensor0/value0` so it isn’t an artifact of Java or Kotlin.

salamanders, 2019-11-21:
Maybe related to https://github.com/ev3dev/ev3dev/issues/62 ?

dlech, 2019-11-21:
Can you give a complete reproducible test case?

salamanders, 2019-11-21:
Absolutely.  Kotlin code to print out gyro angle values for a few seconds.  
Rocking the gyro should produce continual values in Degrees, and spinning the gyro should produce ongoing rotations (rotating past 360 degrees should continue with 361, rotating less than 0 should continue with -1).  Instead, the output is always 0..255, rocking it "backwards" suddenly wraps around to 255.  Almost like it is only one bite of data.

You can see a lot of the same angle data graphed [at this sheet](https://docs.google.com/spreadsheets/d/1VTPyiQIeo71Cw6UfT2cIGR_BHopeKY8BzKksNEo2Ii8/edit#gid=0), including the "wrap at 255" behavior.  

I haven't yet tested to see if 360 degree turn in real life = 256 of these units, but that would be my guess.

Also, AngleRate wasn't working, so maybe I'm configuring the Gyro wrong?  I bet that reset also changes the mode to "angle only".  But still, the degrees should work.

```kotlin
fun main() {
    println("Connecting to Gyro, hold still until numbers start")
    val sensorGyro = EV3GyroSensor(SensorPort.S1)
    sensorGyro.currentMode = 2
    val sampleProvider = sensorGyro.angleAndRateMode
    val sample = FloatArray(sampleProvider.sampleSize())
    sensorGyro.reset() // I'm thinking this isn't needed, that it calibrates automatically when started?
    val endTime = System.currentTimeMillis() + 5_000
    while (System.currentTimeMillis() < endTime) {
        sampleProvider.fetchSample(sample, 0)
        println(sample[0])
        Delay.msDelay(100)
    }
}
```

dlech, 2019-11-23:
I am not able to reproduce the problem. I used the following since I didn't want to setup a java environment.

```python
#!/usr/bin/env python3

from time import sleep

from ev3dev2.sensor.lego import GyroSensor
from ev3dev2.sensor import INPUT_3

gyro = GyroSensor(INPUT_3)

while True:
    print(gyro.angle_and_rate)
    sleep(1)
```

Output looks like this:

```
 $: ./issue1351.py 
(0, 0)
(11, 34)
(48, 51)
(115, 107)
(206, 52)
(269, 85)
(351, 7)
(436, 88)
(409, -158)
(254, -290)
(60, -182)
(-90, -195)
(-202, -129)
(-318, -115)
(-366, 55)
(-236, 55)
(-221, 7)
(-247, -18)
(-252, 22)
(-245, -55)
(-247, -7)
```

As you can see, the angle (first number) is not limited to 0 to 255.

What is the date code printed on your gyro sensor? e.g. 26N5

salamanders, 2019-11-24:
23N6.  Let me try your way.

dlech, 2019-11-24:
I missed the BrickPi3 part earlier. This is probably a BrickPi3 firmware bug, but I will look more tomorrow.

salamanders, 2019-11-24:
Cool! ... Anything I can do to confirm?

dlech, 2019-11-24:
What is the output of `ev3dev-sysinfo -m` on your Pi?

dlech, 2019-11-24:
Well, apparently I'm staying up late so...

I've confirmed the problem on the BrickPi3. There is a mode on the EV3 Color sensor that uses the same data format and that is working, so I am quite confident this is a BrickPi3 firmware issue.

Best thing to do would be to test using Raspbian 4 Robots and if the same problem is there, file an issues with Dexter Industries.

salamanders, 2019-11-24:
Thank you for confirming!
... how odd! 
Posted on https://forum.dexterindustries.com/t/gyro-lego-ev3-gyro-and-256-instead-of-degrees/7035