ev3dev/ev3dev #1353: python3-ptvsd fails to install on buster

dlech, 2019-11-24:
Not sure if this is the same issue: Same package, similar errors.
BrickPi3, raspberrypi, buster.

```
robot@ev3dev:~$ sudo apt install linux-image-ev3dev-rpi2
[sudo] password for robot: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
linux-image-ev3dev-rpi2 is already the newest version (34).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up python3-ptvsd (5.0.0~a5-1ev3dev1) ...
  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_SimpleXMLRPCServer.py", line 263
    except Fault, fault:
                ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_SocketServer.py", line 339
    print '-'*40
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('-'*40)?

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_inspect.py", line 293
    raise TypeError, 'arg is a built-in module'
                   ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_pkgutil_old.py", line 543
    except IOError, msg:
                  ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydev_imps/_pydev_xmlrpclib.py", line 294
    True, False = True, False
    ^
SyntaxError: can't assign to keyword

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydevd_bundle/pydevconsole_code_for_ironpython.py", line 105
    except SyntaxError, err:
                      ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/_pydevd_bundle/pydevd_exec.py", line 3
    exec exp in global_vars, local_vars
           ^
SyntaxError: Missing parentheses in call to 'exec'

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydev_ipython/inputhookglut.py", line 98
    print '\nKeyboardInterrupt'
                              ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('\nKeyboardInterrupt')?

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/_check.py", line 2
    print add_code_to_python_process.run_python_code(3736, "print(20)", connect_debugger_tracing=False)
                                   ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/winappdbg/plugins/do_example.py", line 37
    print "This is an example command."
                                      ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("This is an example command.")?

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/winappdbg/plugins/do_exchain.py", line 40
    print "Exception handlers for thread %d" % thread.get_tid()
                                           ^
SyntaxError: invalid syntax

  File "/usr/lib/python3/dist-packages/ptvsd/_vendored/pydevd/pydevd_attach_to_process/winappdbg/plugins/do_exploitable.py", line 46
    print "-" * 79
            ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("-" * 79)?

dpkg: error processing package python3-ptvsd (--configure):
 installed python3-ptvsd package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 python3-ptvsd
E: Sub-process /usr/bin/dpkg returned an error code (1)
robot@ev3dev:~$

_Originally posted by @salamanders in https://github.com/ev3dev/ev3dev/issues/1198#issuecomment-557849613_

dlech, 2019-11-24:
Should be fixed in `ptvsd_5.0.0~a5-1ev3dev2`

dlech, 2019-11-24:
https://github.com/ev3dev/ptvsd/commit/0ebbb68c3266a2871e00fe295fa465c0262bc630

closed 2019-11-24 by dlech

salamanders, 2019-11-24:
> ev3dev/ptvsd@0ebbb68

Fantastic! Is it easy for me to get this fix, or if not (and often not)
about how long before it shows up in main and I can try again to confirm?

dlech, 2019-11-24:
sudo apt update
    sudo apt upgrade

salamanders, 2019-11-24:
Cool, I was doing `sudo apt-get update && sudo apt-get upgrade` and your way worked!