ev3dev/ev3dev #1354: Ev3 working with rcx sensors

JavitoLope, 2019-11-24:
I read this: "If someone wants design a new converter cable, we could make them work." in
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html

I made this converter cable:

```
P5___
P1___|___(+)
P3_______(-)
```

I'd like to unload ev3dev sensors driver with modprobe and compile this pseudocode program:

```
void main() {
  config(P5,output);
  put(P5,1);
  wait(300);
  config(P5,input);
  wait(0.1);
  print get(P1);
}
```

Any help with program?

Thanks for this great software.

dlech, 2019-11-24:
I'm not sure that P5 would have enough current to actually power active RCX sensors. But it should be possible to enable battery voltage on P1 similar to NXT. (For anyone else reading this, we use P3 on EV3 instead of P2 like NXT because on NXT P2 and P3 and ground, but on EV3, P2 is not ground).

So the wiring would just look like this:

```
P1_______(+)
P3_______(-)
```

For a passive device, like a touch sensor. We can just force the port into `nxt-analog` mode and measure the voltage on P1. `echo nxt-analog > /sys/class/lego-port/portX/mode`. A generic NXT analog sensor will appear in `/sys/class/lego-sensor/` that can be used to read 0-5V.

For an active device, we need to put the port into `raw` mode. `echo raw > /sys/class/lego-port/portX/mode`. Then there will be gpios available in `/sys/class/gpio/gpioX/*`. Then we need to look at [the CPU pinout](https://www.ev3dev.org/docs/kernel-hackers-notebook/ev3-processor/) to figure out which gpio is which. For example "Input Port 3 Pin 1 Out" is "GPIO8-9" which translates to 16*8+9 = gpio137.

The pseudo code look fine to me other than replacing P5 with P1 (although it has been a while, I don't remember the timing off hand of how long the power is supposed to be turned on). There are `direction` and `value` attributes in the gpio sysfs that I mentioned above that will cover most of the functions.

Unfortunately, it looks like we forgot to allow changing the direction of the gpio on P1 so the `direction` attribute is missing from that gpio in the current ev3dev kernels, but we can fix that.

Then to read the analog voltage, we us `/sys/bus/iio/devices/iio:device0/`. See the [wiki](https://github.com/ev3dev/ev3dev/wiki/Internals:-ev3dev-stretch#accessing-the-raw-analog-values) for more info.

dlech, 2019-11-24:
> Unfortunately, it looks like we forgot to allow changing the direction of the gpio on P1 so the `direction` attribute is missing from that gpio in the current ev3dev kernels, but we can fix that.

Actually, I take this back. The GPIO on pin 1 just turns the batter voltage on or off, so it is always output and never input, so just set it high or low.

JavitoLope, 2019-11-24:
I'll try this script with a rcx rotation sensor connected to input port 1

```
echo raw > /sys/class/lego-port/port0/mode
echo 1 > /sys/class/gpio/gpio138/value
sleep 0.3
echo 0 > /sys/class/gpio/gpio138/value
sleep 0.001
cat /sys/bus/iio/devices/iio\:device0/in_voltage6_raw
```

Thank you very much David.

JavitoLope, 2019-12-07:
Should I unload some driver? Thanks.

dlech, 2019-12-07:
The gpio gets automatically exported when the input port is placed in `raw` mode, so ` sudo bash -c 'echo 137 >/sys/class/gpio/export'` is not necessary.

dlech, 2019-12-07:
Once the port is in `raw` mode, `ls /sys/class/gpio` should show several `gpioX` nodes where `X` is a number.

JavitoLope, 2019-12-07:
It works perfect, sorry I forgot the raw change before looking at gpio. Thank you so much.

JavitoLope, 2020-01-14:
I connected a light rcx sensor. Sensor red light blinks and adc always returns 502/503. I think I can not get timming right using shell script.

JavitoLope, 2020-01-15:
Maybe pwm? I don't know what to do. It should be great to have a "rcx" mode in /sys/class/lego-port/portX/modes