ev3dev/ev3dev #1357: Updating screen

JavitoLope, 2019-12-07:
I'm using ev3dev-stretch-ev3-generic-2019-10-23 image and trying to display a picture.

1) I stopped brickman
2) cat /dev/urandom > /dev/fb0
    Nothing is showed in display
3) fbgrab screen.bmp

screen.bmp constains the random image, but display shows nothing.

dlech, 2019-12-07:
1. There is no need to stop `brickman`. If you need run your program from the command line, use `brickrun -- ./my-program`.
2. In ev3dev-stretch, the `/dev/fb0` device has to be `mmap`ed. Regular file read and write don't work.

JavitoLope, 2019-12-07:
Thanks David. Before ev3dev-stretch-ev3-generic-2019-10-23, I tried ev3dev-jessie-ev3-generic-2017-09-14 and cat to display works, then I tried a raspberry pi with raspbian buster and cat to display works. I have installed a lot of software in stretch, but if I use snapshot-ev3dev-buster-ev3-generic-2019-10-23.img.xz will regular file write work?

dlech, 2019-12-07:
Whether `cat` works or not depends on the display driver in the Linux kernel. ev3dev-buster still uses the same driver as ev3dev-stretch so it also requires `mmap`.

JavitoLope, 2019-12-07:
Which is the reason? Is it possible to change the ev3 display driver? I've seen ev3 python source code and it is using nmap like you say, but old versions worked with w/r files (and in this old 1bpp example https://www.ev3dev.org/docs/tutorials/using-ev3-lcd/)

dlech, 2019-12-07:
The current driver is in the mainline Linux kernel, so it is unlikely to change. The use of fbdev devices is trying to be phased out in Linux.

JavitoLope, 2019-12-08:
Finally I changed dotnet code to use mmap and now it's working perfect. Thank you so much David.

JavitoLope, 2019-12-09:
Now, I am writing a text mode program, but it shows weird characters in display, using ssh works ok.

It can be replicated with scripts:

display capture:
[image]

ssh capture
[image]

Is there any way to configure it?

dlech, 2019-12-09:
There is a `setfont` command that can be used to select the font on the display. Note that `setfont` has to be run on the virtual terminal that is being displayed. Available fonts are in `/usr/share/consolefonts`. Most fonts DO NOT use Unicode text encoding, so the program that is printing to the display has to use text encoding that matches the selected console font, otherwise it doesn't look right.

JavitoLope, 2019-12-09:
Thanks, now I see. Where did you get Lat15-TomThumb4x6? Did you create it?

dlech, 2019-12-09:
http://robey.lag.net/2010/01/23/tiny-monospace-font.html

JavitoLope, 2019-12-09:
Now working perfect:

[image]

Again thank you so much David.

closed 2020-01-16 by JavitoLope