ev3dev/ev3dev #1359: RPyC connection to EV3 host fails when an attempt to connect is made

lgut, 2019-12-14:
**What are you trying to do?**

Follow the Rpyc tutorial for ev3dev located here: https://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/rpyc.html

**What did you expect to happen**

I expected to follow the steps of the tutorial and get the same implied results.

**What actually happened**

The program fails when a connection is made. 

On the server/slave side of the connection, the process gets this far before an error is encountered.
I've pasted the full traces at the end of this post as they are quite long.

```
INFO:SLAVE/18812:server started on [0.0.0.0]:18812
INFO:SLAVE/18812:accepted 10.0.0.206:57640
INFO:SLAVE/18812:welcome [10.0.0.206]:57640
DEBUG:SLAVE/18812:Exception caught
```

These failure happen when I use:
- The same version of python on both machines, 3.5.3
- Different versions of python, 3.7.4 on pc, 3.5.3 on ev3
- On all wifi networks I've tried on
- On a Bluetooth PAN using my laptop as an access point (OS and version below)
- With and without a firewall enabled on the laptop

I also followed the suggestion here https://github.com/ev3dev/ev3dev/issues/1031#issuecomment-532500943 and it did not have an effect, which makes sense since the server is bound to 0.0.0.0

My laptop and desktop are able to communicate over rpyc just fine. So far, I can only conclude that there is something not working properly on the ev3.

**Source Code**

Here is the program I am trying to run.

```
import rpyc

conn_ip = '10.0.0.54'

print("attempt connect")
conn = rpyc.classic.connect(conn_ip) # exception is thrown here
print("connected")
```

**Desktop (please complete the following information):**
 - OS and version: 4.19.84-1-MANJARO
 - Connection Type: WiFi, Bluetooth PAN
 - Software: Visual Studio Code, ssh, python repl

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v1.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534FC9D0
BOARD_INFO_TYPE=main
```
[... 177 lines cut ...]
    self._conn.serve(self._ttl)
  File "/home/luis/.pyenv/versions/3.5.3/lib/python3.5/site-packages/rpyc/core/protocol.py", line 387, in serve
    self._dispatch(data)
  File "/home/luis/.pyenv/versions/3.5.3/lib/python3.5/site-packages/rpyc/core/protocol.py", line 355, in _dispatch
    obj = self._unbox(args)
  File "/home/luis/.pyenv/versions/3.5.3/lib/python3.5/site-packages/rpyc/core/protocol.py", line 301, in _unbox
    proxy = self._netref_factory(id_pack)
  File "/home/luis/.pyenv/versions/3.5.3/lib/python3.5/site-packages/rpyc/core/protocol.py", line 314, in _netref_factory
    cls_methods = self.sync_request(consts.HANDLE_INSPECT, id_pack)
  File "/home/luis/.pyenv/versions/3.5.3/lib/python3.5/site-packages/rpyc/core/protocol.py", line 464, in sync_request
    return self.async_request(handler, *args, timeout=timeout).value
  File "/home/luis/.pyenv/versions/3.5.3/lib/python3.5/site-packages/rpyc/core/async_.py", line 102, in value
    raise self._obj
KeyError: ('3048351248', 'SlaveService', 'rpyc.core.service')

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/usr/lib/python3/dist-packages/rpyc/core/protocol.py", line 539, in _handle_inspect
    return tuple(netref.inspect_methods(self._local_objects[oid]))
  File "/usr/lib/python3/dist-packages/rpyc/lib/colls.py", line 86, in __getitem__
    return self._dict[key][0]
KeyError: ('3048351248', 'SlaveService', 'rpyc.core.service')
```

dwalton76, 2019-12-30:
> But rpyc allows for simple communication between devices without the need of a broker or a connection to the outside internet which are qualities I like in a classroom setting

FYI MQTT does not require connection to the internet, the broker can be on the local network (it can be an EV3).

Regarding the error you are hitting check to see if you are running the same version of RPyC on your client and server...just a hunch based on reading [this](https://github.com/tomerfiliba/rpyc/issues/263).

matsibej, 2020-12-24:
> FYI MQTT does not require connection to the internet, the broker can be on the local network (it can be an EV3).
>
> Regarding the error you are hitting check to see if you are running the same version [...]

To check the version      **import rpyc**    then  check the value of   '**rpyc.\_\_version\_\_**'

To upgrade to a new version 
uninstall:                       **pip uninstall rpyc**
install specific version:  **pip install rpyc==4.1.5**

obo, 2021-01-01:
I am probably facing the same issue with freshly flashed ``4.14.117-ev3dev-2.3.5-ev3`` (ev3dev-stretch-ev3-generic-2020-04-10.zip).

@matsibej Does your suggestion apply to the rpyc on EV3 (server) or desktop (client)?

My versions are:
EV3:
```
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
...
>>> rpyc.__version__
(3, 3, 0)
```

desktop:
```
Python 3.5.3 (default, Jan  1 2021, 15:15:01) 
[GCC 5.4.0 20160609] on linux
...
>>> rpyc.__version__
(5, 0, 0)
```

How do I upgrade on EV3 where there is no pip/pip3 pre-installed? (I am always furious from python's zillion of ways of getting installed and getting modules installed, I constantly keep running into incompatibilities.)

obo, 2021-01-01:
I installed upgraded rpyc on EV3 to (5, 0, 0) using these commands:

```
sudo apt-get install python3-pip
  # took very long
sudo pip3 install rpyc

# check that python3 rpyc is now (5, 0, 0)

sudo reboot
  # to hope that rpyc service set up according to the tutorial will restart and use the fresh rpyc
```

I am still unable to connect rpyc to EV3:

On desktop:
```
Python 3.5.3 (default, Jan  1 2021, 15:15:01) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rpyc
>>> ip='192.168.0.105'
>>> conn = rpyc.classic.connect(ip)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/utils/classic.py", line 76, in connect
    return factory.connect(host, port, SlaveService, ipv6=ipv6, keepalive=keepalive)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/utils/factory.py", line 99, in connect
    return connect_stream(s, service, config)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/utils/factory.py", line 56, in connect_stream
    return connect_channel(Channel(stream), service=service, config=config)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/utils/factory.py", line 44, in connect_channel
    return service._connect(channel, config)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/service.py", line 105, in _connect
    self.on_connect(conn)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/service.py", line 213, in on_connect
    self._install(conn, conn.root)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 506, in root
    self._remote_root = self.sync_request(consts.HANDLE_GETROOT)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 473, in sync_request
    return self.async_request(handler, *args, timeout=timeout).value
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/async_.py", line 100, in value
    self.wait()
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/async_.py", line 47, in wait
    self._conn.serve(self._ttl)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 396, in serve
    self._dispatch(data)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 364, in _dispatch
    obj = self._unbox(args)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 298, in _unbox
    proxy = self._netref_factory(id_pack)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 312, in _netref_factory
    cls_methods = self.sync_request(consts.HANDLE_INSPECT, id_pack)
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/protocol.py", line 473, in sync_request
    return self.async_request(handler, *args, timeout=timeout).value
  File "/home/bojar/notes/lego/s3parator/testing-rpyc/venv353/lib/python3.5/site-packages/rpyc/core/async_.py", line 102, in value
    raise self._obj
KeyError: ('3048278128', 'SlaveService', 'rpyc.core.service')

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/usr/lib/python3/dist-packages/rpyc/core/protocol.py", line 539, in _handle_inspect
    return tuple(netref.inspect_methods(self._local_objects[oid]))
  File "/usr/lib/python3/dist-packages/rpyc/lib/colls.py", line 86, in __getitem__
    return self._dict[key][0]
KeyError: ('3048278128', 'SlaveService', 'rpyc.core.service')
```

What should I check on EV3?

skyaee, 2021-01-16:
I have the same problem. ev3dev has been updated to rpyc 5.0.0, but there is a problem when starting the service.

robot@ev3dev:~$ python3
Python 3.5.3 (default, Nov 18 2020, 21:09:16)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rpyc
>>> rpyc.version.version
(5, 0, 0)
>>>
robot@ev3dev:~$ which rpyc_classic.py
/usr/bin/rpyc_classic.py
robot@ev3dev:~$ python3 ^C
robot@ev3dev:~$ python3 /usr/bin/rpyc_classic.py
Traceback (most recent call last):
  File "/usr/bin/rpyc_classic.py", line 125, in <module>
    ClassicServer.run()
  File "/home/robot/.local/lib/python3.5/site-packages/plumbum/cli/application.py", line 572, in run
    retcode = inst.main(*tailargs)
  File "/usr/bin/rpyc_classic.py", line 85, in main
    self._serve_mode(ThreadedServer)
  File "/usr/bin/rpyc_classic.py", line 96, in _serve_mode
    registrar = self.registrar, auto_register = self.auto_register)
  File "/home/robot/.local/lib/python3.5/site-packages/rpyc/utils/server.py", line 83, in __init__
    address = socket.getaddrinfo(hostname, port, family=family, type=socket.SOCK_STREAM, proto=socket.IPPROTO_TCP, flags=socket.AI_PASSIVE)[0][-1]
  File "/usr/lib/python3.5/socket.py", line 733, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

pbackx, 2021-02-01:
@skyaee I had the same issue. Installed 5.0.0 and got the "name or service not known" error.

I was able to fix this by explicitly defining the host with the "--host" parameter.

So in the /lib/systemd/system/rpyc-classic.service file, you modify this line:

```ExecStart=/usr/bin/rpyc_classic.py --host <ip of your ev3>```

snaildos, 2021-03-30:
Still getting this problem. -=-

tendwong, 2022-07-23:
Find out what is the version of Python and RPYC you have on your EV3DEV. For mine, the Python version is 3.5.3 and RPYC is 3.3.0. To fix the problem, you have to install the same version of Python and same version of RPYC on your desktop.

Python 3.5.3
https://www.python.org/downloads/release/python-353/

If you are on Windows, type 'py -m pip list' to see what RPYC you have. To install version 3.3.0 of RPYC, type 'py -m pip install rpyc==3.3.0'

gjhoppy, 2024-02-06:
I found I needed to add the ip address to rpyc-classic.service file on the brick. Default is meant to be 0.0.0.0 but without it connection is refused.  Later versions test on PC shows loopback being used.

rpyc-classic.service
ExecStart=/usr/bin/rpyc_classic.py --host "0.0.0.0"

Also need to run RPYC version 3.3.0 on desktop. Python versions can differ but 3.6 is the last version that works.

maalsol, 2024-07-20:
> I found I needed to add the ip address to rpyc-classic.service file on the brick. Default is meant to be 0.0.0.0 but without it connection is refused. Later versions test on PC shows loopback being [...]

Growflavor, 2024-07-24:
> > I found I needed to add the ip address to rpyc-classic.service file on the brick. 
> > rpyc-classic.service ExecStart=/usr/bin/rpyc_classic.py --host "0.0.0.0"
> 
> I struggled for a long time with this, but I can confirm now that you don't need rpyc 3.3.0 on desktop. I currently have 5.0.0 on both the EV3 brick ( with python 3.5.3) and in my desktop (with python 3.11.3).

Yes, on EV3 Debian Stretch one can sudo python3 -m pip install rpyc==5.0.0 which will connect with RPyC 5.3.1 in JupyterNotebook on the desktop (even though 5.0.0 was yanked since it is not supposed to install on Stretch).

https://github.com/ev3dev/ev3dev/issues/1554#issuecomment-2019205120

Also, the experimental Debian Buster #1634 & Bullseye #1636 EV3 images will pip install RPyC 6.x by default or can be told to sudo python3 -m pip install rpyc==5.3.1