ev3dev/ev3dev #1360: Pip3 extremely slow - keeps hanging

faizz, 2019-12-20:
I am tying to update the version of ev3dev on Debian using pip3. I then intend to install some machine learning packages.
However even running "pip3 --version" takes longer than 5 minutes. The result of this command is:
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)

After letting "pip3 install python-ev3dev2" run for about 10-15 minutes the following is the result:
ConnectionResetError(104, 'Connection reset by peer'))

I am using a standard EV3 brick with a Netgear WG111V2 USB Wireless adaptor

Here is the log file:
[https://gist.github.com/faizz/d552e8137b4b2f0d92ff9ed83984486f](url)

WasabiFan, 2019-12-26:
I don't have any great ideas but you could check `top` to see if something is eating CPU. It might be the Wi-Fi dongle driver. Also, perhaps your SD card is faulty or otherwise incompatible -- you could try swapping it out.

dwalton76, 2019-12-30:
It takes almost 4 minutes for me
```
robot@ev3dev:~$ time pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.5)

real	3m48.159s
user	2m45.690s
sys	0m4.755s
robot@ev3dev:~$ 
```

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-12-14
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533F8FD1
BOARD_INFO_TYPE=main
```

WasabiFan, 2020-01-01:
That's funny. I wonder why it takes that long... What does `top` show while it's churning?

dwalton76, 2020-01-01:
nothing crazy
```

top - 14:10:05 up 4 min,  2 users,  load average: 2.05, 2.13, 1.00
Tasks:  77 total,   2 running,  44 sleeping,   0 stopped,   0 zombie
%Cpu(s): 80.9 us, 18.8 sy,  0.0 ni,  0.0 id,  0.0 wa,  0.0 hi,  0.3 si,  0.0 st
KiB Mem :    57620 total,     3520 free,    36604 used,    17496 buff/cache
KiB Swap:    98300 total,    89084 free,     9216 used.    17340 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND                     
 1028 robot     20   0   21688  16368   6484 R 65.5 28.4   0:48.28 pip3                        
   20 root      20   0       0      0      0 S  8.1  0.0   0:04.16 kswapd0                     
   89 root     -51   0       0      0      0 S  7.5  0.0   0:18.70 irq/249-ev3-tac             
  973 robot     20   0    8084   1828   1368 R  6.5  3.2   0:06.76 top                         
   68 root     -51   0       0      0      0 S  4.3  0.0   0:12.11 irq/247-ti-ads7             
  501 root      20   0   49740   2828   1992 S  1.6  4.9   0:10.13 brickman 
```

WasabiFan, 2020-01-04:
65% means it _is_ doing something, rather than just getting stuck. I am not sure how to easily figure out what it's doing, other than maybe <kbd>Ctrl</kbd>+<kbd>C</kbd> while it's working and take a look at the stack trace.