ev3dev/ev3dev #1362: EVB 1.0 Won't Boot with BBB

Xaplomian, 2019-12-28:
**What are you trying to do?**

I am trying to get the EVB to boot.

**What did you expect to happen**

I expected it to boot.

**What actually happened**

It just displayed a white screen. All the LEDs on the BBB did turn on though: power and all four user LEDs.

**Robot (please complete the following information):**
 - Device: EVB with BBB
 - Versions: ev3dev-stretch R1 and ev3dev-stretch R2
 - I have installed the normal firmware for EVB with the SD card to confirm that the hardware work and SD card work.

**Background Information**

I just want to use the EVB.

dlech, 2019-12-28:
Did you edit the `uEnv.txt` on the SD card file first?

Xaplomian, 2019-12-28:
I did not, as I was not aware that was necessary. It is now working.

closed 2019-12-28 by dlech