ev3dev/ev3dev #1363: Brickman menu is unreadable

Krysek, 2019-12-28:
**What are you trying to do?**
I try to prepare my Raspberry Pi with PiStorms V2 for EV3 developments (C/C++).
I flashed and adjust a SD card for the Raspberry Pi with the ev3dev image from source "https://www.ev3dev.org/downloads/" according to the manual from "http://docs.ev3dev.org/en/ev3dev-stretch/getting-started/pistorms.html".

Unfortunately, the Brickman menu is unusable: I can't see any text!
 (my ideas: No text are displayed or texts are vanished due to the small font size.)

In general, my Pi works:
   - SSH works.
   - I can steer Lego motors via C++.
   - I can connect and process data from my Pixy cam via C++.
   - Sound output works (espeak).

I am trying to ...?
1st: I tried to use a keyboard to check if the menu reacts on any input: Yes, it reacts. I could go up and down within a menu list. 

2nd: I search on the internet and here and amongst other I found this issue: "How to adjust font size in brickman?" #1039 
I tried out all hints, but nothing has changed. I don't still see any text.

Perhaps, I read an important detail over.

**What did you expect to happen**
I'd like to see text within Brickman menu.

**What actually happened**
Nothing has changed. I don't still see any text within Brickman menu.

**Screenshots**
Brickman menu:
[image]

**Robot (please complete the following information):**
Flashed image: ev3dev-stretch-rpi2-generic-2019-10-23.img

```
Image file:         ev3dev-stretch-rpi2-generic-2019-10-23
Kernel version:     4.14.114-ev3dev-2.3.4-rpi2
Brickman:           0.10.2
BogoMIPS:           38.40
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000d08d05cf
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=V2.08
BOARD_INFO_MODEL=PiStorms
BOARD_INFO_TYPE=aux
```

I hope, someone is able to help me.

Cheers,
Krysek

dlech, 2019-12-28:
The picture is not very clear, but do I see a Wi-Fi symbol at the top?

Also, in an SSH terminal, try this and paste the output here:

```
sudo systemctl stop brickman
sudo G_MESSAGES_DEBUG=all brickman
```

Krysek, 2019-12-28:
You are right. At the top is the Wi-Fi symbol.

Here a bigger pic:
[image]

... and here my the output of G_MESSAGES_DEBUG=all brickman.

```
$ sudo G_MESSAGES_DEBUG=all brickman
** (process:9200): DEBUG: Mode: 64
** (process:9200): DEBUG: GRX_PLUGIN_PATH=/usr/lib/arm-linux-gnueabihf/grx-3.0/plugins/2
** (process:9200): DEBUG: Checking GRX_DRIVER environment variable: (null)
** (process:9200): DEBUG: Trying detect
** (process:9200): DEBUG: Looking for plugin: /usr/lib/arm-linux-gnueabihf/grx-3.0/plugins/2/libgrx-3.0-vdriver-gtk3.so
** (process:9200): DEBUG: /usr/lib/arm-linux-gnueabihf/grx-3.0/plugins/2/libgrx-3.0-vdriver-gtk3.so: cannot open shared object file: No such file or directory
** (process:9200): DEBUG: Looking for plugin: /usr/lib/arm-linux-gnueabihf/grx-3.0/plugins/2/libgrx-3.0-vdriver-linuxfb.so
** (process:9200): DEBUG: Looking for plugin: /usr/lib/arm-linux-gnueabihf/grx-3.0/plugins/2/libgrx-3.0-vdriver-memory.so
** (process:9200): DEBUG: /usr/lib/arm-linux-gnueabihf/grx-3.0/plugins/2/libgrx-3.0-vdriver-memory.so: cannot open shared object file: No such file or directory
** (process:9200): DEBUG: Found linuxfb
** (process:9200): DEBUG: Attempting to set mode in linuxfb
** (process:9200): DEBUG: Mode 79 txt/grx mismatch
** (process:9200): DEBUG: Mode 78 OK!
** (process:9200): DEBUG: buildcontext - Mode Frame buffer = 0x75f33000
** (process:9200): DEBUG: buildcontext - Mode Frame selector = 0x76d6f4c0
** (process:9200): DEBUG: plane size: 153600
** (process:9200): DEBUG: buildcontext - Linear Mode
** (process:9200): DEBUG: buildcontext - context buffer 0 = 0x75f33000
** (process:9200): DEBUG: buildcontext - context buffer 1 = 0x75f33000
** (process:9200): DEBUG: buildcontext - context buffer 2 = 0x75f33000
** (process:9200): DEBUG: buildcontext - context buffer 3 = 0x75f33000
** (process:9200): DEBUG: grx_color_info_reset_colors ...
** (process:9200): DEBUG: grx_color_info_reset_colors done
** (process:9200): DEBUG: grx_set_mode complete
** (process:9200): DEBUG: _GrInitMouseCursor ...
** (process:9200): DEBUG: _GrInitMouseCursor done
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported

** (brickman:9200): CRITICAL **: grx_text_options_new_full: assertion 'font != NULL' failed
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported

** (brickman:9200): CRITICAL **: grx_text_options_new_full: assertion 'font != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_set_font: assertion 'options != NULL' failed
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported

** (brickman:9200): CRITICAL **: grx_text_options_new_full: assertion 'font != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_set_font: assertion 'options != NULL' failed
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported
** (brickman:9200): DEBUG: searching for pattern: fixed sans-8:slant=0:weight=80:width=100:outline=False:scalable=False:dpi=162:fontformat=PCF

** (brickman:9200): CRITICAL **: Fonts.vala:42: Only bitmap fonts are supported

** (brickman:9200): CRITICAL **: grx_text_options_new_full: assertion 'font != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_set_font: assertion 'options != NULL' failed
[... 2916 lines cut ...]

** (brickman:9200): CRITICAL **: grx_text_options_get_v_align: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_draw_text: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_get_font: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_font_get_height: assertion 'font != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_set_fg_color: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_get_font: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_font_get_text_width: assertion 'font != NULL' failed

** (brickman:9200): CRITICAL **: grx_draw_text: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_set_fg_color: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_text_options_get_font: assertion 'options != NULL' failed

** (brickman:9200): CRITICAL **: grx_font_get_text_width: assertion 'font != NULL' failed

** (brickman:9200): CRITICAL **: grx_draw_text: assertion 'options != NULL' failed
```

dlech, 2019-12-28:
Cool, thanks. It looks like we messed up fonconfig somehow. (This issue is duplicate of #1339).

Try `sudo rm /etc/fonts/conf.d/70-no-bitmaps.conf`.

Krysek, 2019-12-28:
Thanks! Now I can read all options!

closed 2019-12-28 by dlech