ev3dev/ev3dev #1366: AM1808 IPU/IPD

JavitoLope, 2020-01-09:
How can I enable IPU?

dlech, 2020-01-09:
What is IPU/IPD?

JavitoLope, 2020-01-09:
AM1808 Microporcessor's internal pullup/pulldown resistors. I´m working with a sensor, it sends pulses at irregular intervals and I need to count them, so I thought using digital inputs putting the ev3 port in raw mode.

dlech, 2020-01-09:
The EV3 uses external pullup/pulldown resistors everywhere, so all internal pullup/pulldown on the SoC are disabled. The EV3 schematics are available for download on the LEGO website.

JavitoLope, 2020-01-09:
I read they are disabled at https://www.ev3dev.org/docs/kernel-hackers-notebook/ev3-processor/. The idea is to enable IPU on DIGIAI0 or DIGIAI1, I know how to setup an IPU in raspberry pi with /dev/gpiomem and mmap, but I don´t know how to do this with ev3 hardware.

dlech, 2020-01-09:
DIGIAI0 and DIGIAI1 are not connected directly to the SoC. There are series resistors between them and the SoC, so if you enable the internal pull up, the GPIO will probably always read high.

We experienced this during initial development where one of the buttons on the EV3 wasn't working because the internal pullup was not disabled. It thought the button was always pressed no matter what.

DIGIAI0 is already pulled high and DIGIAI1 is already pulled low with external resistors. I'm afraid it can't be changed.

JavitoLope, 2020-01-09:
I downloaded the schematics you told me and now I can see the external resistors. With them, it is not necessary to change IPU. I also need to connect an arduino i2c slave, it works with 5V (https://www.dexterindustries.com/howto/connecting-ev3-arduino/). Do you think I can connect arduino i2s 5V to DIGIAI0 and DIGIAI1?

dlech, 2020-01-09:
It is safe to use 5V.

JavitoLope, 2020-01-09:
Thank you so much David.

dlech, 2020-01-09:
Also when using I2C, the EV3 uses 82k pull up resistors instead of the usual 10k.

JavitoLope, 2020-01-09:
Do you think I need to add resistors? In the schematic of dexter they say did not have to use resistors. In the lego schematic I can see a 4.7k ohm resistor from 5V (arduino output) to SoC input and calculated 3.8V at that point using spice.

dlech, 2020-01-09:
I know that devices that use 10k pull-up resistors on I2C don't work because it pulls up too strongly. I don't think I've ever tried it without any extra resistors.

JavitoLope, 2020-01-14:
Working perfect arduino and ev3 using i2c. Thank you so much David.

closed 2020-01-16 by JavitoLope