ev3dev/ev3dev #1368: Brickrun not returning stdout

scprotz, 2020-01-14:
Brickrun returns stderr to Visual Studio Code just fine in normal operation when using the Lego Mindstorms plugin for Visual Studio Code.  It does not return stdout.

An easy test is to install VSC + Lego Mindstorms plugin and use the current ev3dev image provided on the Lego site.   

1) Create a "Hello World!" program.  
`import sys
print("Hello World!")
print("Hello World stderr!", file=sys.stderr)`

The second line will print in VSC console, the first will not.

scprotz, 2020-01-14:
So researching this more (and as typical), I found some specific brickrun documentation.  It appears there may be a "-r" flag that could fix this and pass through stdout, but I haven't figured a way to send that parameter in Visual Studio Code.

dlech, 2020-01-14:
You can open an SSH terminal in VS Code by right clicking on the device after it is connected, then type in the command in the terminal. 

There is also an open issue for automating this. https://github.com/ev3dev/vscode-ev3dev-browser/issues/90 if you have any ideas on how it _should_ work.

scprotz, 2020-01-14:
Thanks for the comments.  I currently use ssh.  Looks like same issue is already flagged with [ev3dev/vscode-ev3dev-browser#90](https://github.com/ev3dev/vscode-ev3dev-browser/issues/90) so this one should be merged with it.  I'll continue comments there.