ev3dev/ev3dev #1369: Brickrun LEDs Blinking when trying to do unittest

scprotz, 2020-01-14:
I've been trying to set up some unit tests for a lightweight ev3 micropython API (using ev3fast but written for Micropython).

When I try to do unit tests, they are run from the IDE (Visual Studio Code) through brickrun.  Brickrun blinks the leds while running.  This effectively limits the LED tests I can run as the brightness value keeps changing.

Is there a way to disable that feature so that I can properly control the LEDs even when using brickrun?

Thanks

dlech, 2020-01-14:
Are you changing the `trigger` attribute of the LEDs?

scprotz, 2020-01-14:
Ok. I originally thought the Vala code had it on a thread so it kept calling set_leds..but I read through the code again and it seems like it just calls it once.  So after some trial and error, it appears if I make sure my first call to the LEDs is to set the trigger to "none", and at that point, they seem to work as expected.  (I thought I'd already tried this once and it failed, but tried it again and works as expected right now).

I'll close this one as the workaround is to just set the trigger to "none" before using the LEDs.

closed 2020-01-14 by scprotz