ev3dev/ev3dev #1371: Mindstorms EV3 MicroPython Brick Failed to connect to ev3dev: Time out while waiting for handshake

kjtumminello, 2020-01-26:
I have a Mindstorms brick booted up on the EV3 MicroPython image.  I am trying to connect using USB tethering with ev3dev-browser 1.0.4 using VS Code on a Windows 10 machine.

EV3DEV Device Browser finds the device "ev3dev".  I attempt to connect, spins through "Connecting" for about 20 seconds then fails with error:

"Failed to connect to ev3dev: Timed out while waiting for handshake"

I have connected on two other Windows machines without issue.  The only difference I can see is when machine finds device it lists it as:
"**ev3dev** Ethernet 3"

The other two machines where I've been successful list the device as "Ethernet" and "Ethernet 5"

**Desktop:**
 - OS and version: Window 10 Pro 1903
 - Connection Type: USB cable
 - Software: Visual Studio Code 1.38.1

**Robot:**
 - Device: EV3
 - Versions: ev3micropythonv100sdcardimage
# Changelog for ev3-micropython-v1.0.0-sd-card-image
## v1.0.0 - 2019-03-26
- Initial release

```
Name: ev3dev-browser
Id: ev3dev.ev3dev-browser
Description: Browse for ev3dev devices
Version: 1.0.4
Publisher: ev3dev
VS Marketplace Link: https://marketplace.visualstudio.com/items?itemName=ev3dev.ev3dev-browser
```

**Background Information**

dlech, 2020-01-26:
Tethering is used to share the Internet connection of the EV3 to another device. This is most likely not what you want.

If you don't need Internet access on the EV3, then there is no configuration required at all to connect via USB to the VS Code extension.

kjtumminello, 2020-01-26:
Perhaps I mistated my configuration. I am connecting the EV3 brick to the computer using the USB cord so that I can download MicroPython program from computer to EV3 brick.

kjtumminello, 2020-01-26:
I am using the same robot and USB cord which works on my home computer and my personal laptop (all Windows 10).  I am wanting to teach a class for middle school, and when I try to connect on school computers using the same method, I get the "failed to connect" error.

Plouc314, 2020-02-08:
I have exactly the same problem, (with the same sd card image, the same error in Visual Studio Code, the only difference is that I use ubuntu). I've already searched for tutorials and doc but didn't find anything, so  if somebody could answer to kjtumminello it would be very nice.

kjtumminello, 2020-02-08:
Haven't resolved the issue but was able to work through my Bluetooth issues in the meantime.  Was going to use USB for the four robots I have in classroom to keep it simple.  Tried Bluetooth but turned out to be a mess at first with all the robots named "ev3dev".   Now all robots have a unique name and Bluetooth working OK.  Here are the instructions I consolidated in order to rename robots and use Bluetooth.  With these instructions, you need to one-time find a computer that you have admin privileges and can connect to robot with USB.

Set up network connection to robot.  Need a computer that can connect using USB and have admin privileges.
https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/
Connect to use SSH.  To use VS Code editor, right-click on robot name on EV3DEV Device Browser  to establish terminal connection.
https://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/
Change robot name.
http://www.mow-bots.com/how-to-change-your-ev3dev-hostname/
Connecting via Bluetooth
https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

Plouc314, 2020-02-08:
The Bluetooth connection worked for me,
you've just stopped a big headache ;)
Thank you very much !

physcrowley, 2021-02-24:
I need the USB connection for my students (who work on desktops without Bluetooth), but can't configure the Interent access (due to the lack of admin priviledges) required for the NDIS device. 

However, simply configuring the Internet access for the NDIS device doesn't solve the problem for me neither. With that access granted on my Windows 10 laptop, the USB connection still times out (using both the MicroPython image mentioned above and the ev3dev image from the ev3dev installation page) while the Bluetooth connection works fine - even if a few attempts are required at times.

bryceikeda, 2025-02-12:
For anyone still using EV3 Mindstorms bricks, I've had the same connection issues. Here is what is working for me:

1. Click connect to device in VSCode
2. If a device is found and says the Ethernet # next to ev3dev I can connect fine
3. If a device is not found, I click "I don't see my device," then enter the name 'ev3dev' and the IP address 192.168.0.1
4. Then I go to the Brick and click Wireless and Networks -> Tethering -> Gadget
5. Then I try to connect by clicking the device in VSCode

If it still doesn't connect, make sure you can see 'Wired' popup in "All Network Connections"

If it is popping up, try clicking Wired -> Connect Automatically and then Connect, while reclicking your device in VSCode.