ev3dev/ev3dev #1372: WIKI Documentation Appendix A: Sensor Data EV3 Gyro sensor

tom-eilers-nl, 2020-01-28:
In the documentation Appendix a: Sensor DATA EV3 Gyro Sensor writes:

Commands:
This sensor does not support commands.

This is not true , the EV3-G programming allows the Gyro the be reset.

Go to the angle mode --> write 17 --> wait 25 msec.

Also hold  the robot perfectly still when you do this,
otherwise it will mess up the calibration of the sensor.

dlech, 2020-01-28:
You are correct. However in a technical sense, "commands" for other sensors are sent using the `command` attribute and are independent of the current mode. Resetting the gyro has to be done with the `direct` attribute, so it is a bit different and only works with the angle mode.

Would you like to add a note about this here and make a pull request? https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-buster/sensors/ev3_uart_sensor_defs.c#L264

JakubVanek, 2020-02-23:
Hi @tom-eilers-nl,

do you have more information about this procedure? I'm doing some research in #1387 and I think I don't see the whole picture.

Thank you very much,

Jakub Vanek