ev3dev/ev3dev #1374: Very strange 'growing' segfaults (of things like awk, top)

obo, 2020-02-04:
**What are you trying to do?**

I am trying to run my code which successfully ran a year ago. I wanted to revive and record my LoEV3go (https://github.com/obo/loev3go).

**What did you expect to happen**

The program should have started and worked as it did a year ago.

**What actually happened**

Python web server with threaded tank-style robot started, but crashed on the 'Preview'. I succeeded in creating MWE of just trying python-tk demo which crashes:

```
#!/usr/bin/env python3
from tkinter import *
class Window(Frame):
    def __init__(self, master=None):
        Frame.__init__(self, master)               
        self.master = master
root = Tk()
app = Window(root)
root.mainloop()
```

When run in pdb, this is where it segfaults:

```
08:26 [797] tests$./test_tk_inter.py 
> /home/robot/lego/loev3go/tests/test_tk_inter.py(7)<module>()
-> root = Tk()
(Pdb) s
--Call--
> /usr/lib/python3.5/tkinter/__init__.py(1860)__init__()
-> def __init__(self, screenName=None, baseName=None, className='Tk',
(Pdb) s
> /usr/lib/python3.5/tkinter/__init__.py(1867)__init__()
...
> /usr/lib/python3.5/tkinter/__init__.py(1879)__init__()
-> interactive = 0
(Pdb) s
> /usr/lib/python3.5/tkinter/__init__.py(1880)__init__()
-> self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
(Pdb) s
Segmentation fault
```

During the debugging, various programs started crashing, e.g. `awk`, later `top` (which used to work some hours or days back).

awk crashes sometimes:

```
08:45 [775] ~$echo "JO  NE" | awk '{print;}'
JO	NE
08:46 [774] ~$echo "JO  NE" | awk '{$1="NE"; print;}'
awk: cmd. line:1: fatal error: internal error: segfault
Aborted
```

I tried reinstalling the failing packages, e.g. gawk, python-tk, but nothing ever changed.

**Desktop (please complete the following information):**
 - OS and version: Ubuntu 16.04.4. LTS
 - Connection Type: Wi-Fi
 - Software: xterm

**Robot (please complete the following information):**
 - Device: EV3
 - Ports: no sensors or motors connected during these test, only a USB wifi dongle which always worked.

**System info (from `ev3dev-sysinfo`)** (the system info is also affected by the crashing `awk`):
```
Image file:         ev3dev-stretch-ev3-generic-2018-04-22
Kernel version:     4.14.38-ev3dev-2.0.1-ev3
Brickman:           0.10.0
awk: cmd. line:1: fatal error: internal error: segfault
ev3devKit:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535F74D2
BOARD_INFO_TYPE=main
```

** Hardware Checks **

I checked the two partitions of the SD card on my notebook and there are no errors:

```
...
/dev/mmcblk0p2   3785840   2493144   1105064  70% /media/bojar/EV3DEV_ROOTFS
/dev/mmcblk0p1     49038      7322     41716  15% /media/bojar/EV3DEV_BOOT
09:35 obonb ~$sudo umount /media/bojar/EV3DEV_ROOTFS
09:35 obonb ~$sudo umount /media/bojar/EV3DEV_BOOT
09:35 obonb ~$sudo badblocks -n -v /dev/mmcblk0p1
Checking for bad blocks in non-destructive read-write mode
From block 0 to 49151
Testing with random pattern: Pass completed, 0 bad blocks found. (0/0/0 errors)
09:36 obonb ~$sudo badblocks -n -v /dev/mmcblk0p2
Checking for bad blocks in non-destructive read-write mode
From block 0 to 3878911
Testing with random pattern: Pass completed, 0 bad blocks found. (0/0/0 errors)
```

I tried memtester. Yesterday, it worked up to testing 30 MB of RAM:

```
08:28 [779] ~$sudo memtester 30 1
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 30MB (31457280 bytes)
got  30MB (31457280 bytes), trying mlock ...locked.
Loop 1/1:
  Stuck Address       : ok         
  Random Value        : ok
  Compare XOR         : ok
  Compare SUB         : ok
  Compare MUL         : ok
  Compare DIV         : ok
  Compare OR          : ok
  Compare AND         : ok
  Sequential Increment: ok
  Solid Bits          : ok         
...
```

memtest for 40 MB dies:
```
08:26 [780] ~$sudo memtester 40 1
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 40MB (41943040 bytes)
got  40MB (41943040 bytes), trying mlock ...locked.
Loop 1/1:
  Stuck Address       : testing   4Killed
```

Yes, there is probably not enough memory to allocate 40 MB:

```
08:33 [777] ~$cat /proc/meminfo 
MemTotal:          57180 kB
MemFree:           22828 kB
MemAvailable:      39152 kB
Buffers:            3768 kB
Cached:            14492 kB
SwapCached:          724 kB
Active:            12968 kB
Inactive:           7828 kB
Active(anon):       2028 kB
Inactive(anon):      512 kB
Active(file):      10940 kB
Inactive(file):     7316 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:         98300 kB
SwapFree:          90108 kB
Dirty:                48 kB
Writeback:             0 kB
AnonPages:          2488 kB
Mapped:             9744 kB
Shmem:                 4 kB
Slab:               5796 kB
SReclaimable:       1360 kB
SUnreclaim:         4436 kB
KernelStack:         520 kB
PageTables:          572 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      126888 kB
Committed_AS:      67508 kB
VmallocTotal:     966656 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
```

Memtest for 121 MB (which I saw somewhere) killed the whole brick:

```
08:39 [785] ~$sudo memtester 121 5
[sudo] password for robot: 
memtester version 4.3.0 (32-bit)
Copyright (C) 2001-2012 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 121MB (126877696 bytes)
got  121MB (126877696 bytes), trying mlock ...Killed
```
Now the whole brick got stuck (maybe reboot, maybe never quite booted), showing only brickman startup screen.
After another restart (middle button + ESC button held for reset), brickman again gets stuck in the boot screen but sshd starts, so that I can again connect.

Is this really failing HW? How can I diagnose better?

Thanks, O.

dlech, 2020-02-05:
Try using a new microSD card.

I had two fail recently (on the same day!) that would cause random crashes similar to this.

obo, 2020-02-13:
I confirm it was indeed the SD card. A new one behaves normally. So even a card without recognized bad blocks can non-deterministically kill the machine. Thanks, O.

closed 2020-02-13 by obo