ev3dev/ev3dev #1376: Mindstorms freezes while doing apt-get update [C++ ReadMe tutorial]

Niels1006, 2020-02-09:
**Describe the bug**
Hello, I did the tutorial (https://github.com/ddemidov/ev3dev-lang-cpp) and at point 7 my EV3 freezes at around 90-99%.

 **To Reproduce**
Steps to reproduce the behavior:
- 1. `sudo vim /etc/apt/sources.list`
- 2. Add ```deb http://httpredir.debian.org/debian unstable main contrib non-free
    deb http://security.debian.org/ unstable/updates main contrib non-free
    deb http://archive.ev3dev.org/debian unstable main```
- 3. Do `sudo apt-get update` or `sudo apt-get install gcc-5/unstable`
- 4. My Mindstorms freezes at around 90-99% (I cant use brickman etc..) (More details are coming next)

- Used software: PuTTY (ye makes sense)

**Robot:**
 - Device: EV3
 - Versions: Newest version of ev3dev

What happens when I do `sudo apt-get update`:
```
robot@ev3dev:~$ sudo apt-get update
[sudo] password for robot:
Get:1 http://security.debian.org stretch/updates InRelease [94.3 kB]
Ign:2 http://httpredir.debian.org/debian stretch InRelease
Get:3 http://httpredir.debian.org/debian unstable InRelease [142 kB]
Ign:4 http://security.debian.org unstable/updates InRelease
Hit:5 http://archive.ev3dev.org/debian stretch InRelease
Ign:6 http://archive.ev3dev.org/debian unstable InRelease
Ign:7 http://security.debian.org unstable/updates Release
Ign:8 http://archive.ev3dev.org/debian unstable Release
Ign:9 http://security.debian.org unstable/updates/main all Packages
Ign:10 http://archive.ev3dev.org/debian unstable/main all Packages
Hit:11 http://httpredir.debian.org/debian stretch Release
Ign:12 http://security.debian.org unstable/updates/contrib armel Packages
Ign:13 http://security.debian.org unstable/updates/contrib all Packages
Ign:14 http://archive.ev3dev.org/debian unstable/main armel Packages
Ign:15 http://security.debian.org unstable/updates/main armel Packages
Ign:16 http://security.debian.org unstable/updates/non-free all Packages
Ign:17 http://security.debian.org unstable/updates/non-free armel Packages
Ign:10 http://archive.ev3dev.org/debian unstable/main all Packages
Ign:9 http://security.debian.org unstable/updates/main all Packages
Ign:12 http://security.debian.org unstable/updates/contrib armel Packages
Ign:13 http://security.debian.org unstable/updates/contrib all Packages
Ign:15 http://security.debian.org unstable/updates/main armel Packages
Ign:16 http://security.debian.org unstable/updates/non-free all Packages
Ign:14 http://archive.ev3dev.org/debian unstable/main armel Packages
Ign:17 http://security.debian.org unstable/updates/non-free armel Packages
Ign:9 http://security.debian.org unstable/updates/main all Packages
Ign:12 http://security.debian.org unstable/updates/contrib armel Packages
Ign:13 http://security.debian.org unstable/updates/contrib all Packages
Ign:10 http://archive.ev3dev.org/debian unstable/main all Packages
Ign:15 http://security.debian.org unstable/updates/main armel Packages
Ign:16 http://security.debian.org unstable/updates/non-free all Packages
Ign:17 http://security.debian.org unstable/updates/non-free armel Packages
Ign:14 http://archive.ev3dev.org/debian unstable/main armel Packages
Ign:9 http://security.debian.org unstable/updates/main all Packages
Ign:12 http://security.debian.org unstable/updates/contrib armel Packages
Ign:13 http://security.debian.org unstable/updates/contrib all Packages
Ign:10 http://archive.ev3dev.org/debian unstable/main all Packages
Ign:15 http://security.debian.org unstable/updates/main armel Packages
Ign:16 http://security.debian.org unstable/updates/non-free all Packages
Ign:17 http://security.debian.org unstable/updates/non-free armel Packages
Ign:14 http://archive.ev3dev.org/debian unstable/main armel Packages
Ign:9 http://security.debian.org unstable/updates/main all Packages
Ign:12 http://security.debian.org unstable/updates/contrib armel Packages
Ign:13 http://security.debian.org unstable/updates/contrib all Packages
Ign:10 http://archive.ev3dev.org/debian unstable/main all Packages
Ign:15 http://security.debian.org unstable/updates/main armel Packages
Ign:16 http://security.debian.org unstable/updates/non-free all Packages
Ign:17 http://security.debian.org unstable/updates/non-free armel Packages
Ign:14 http://archive.ev3dev.org/debian unstable/main armel Packages
Ign:9 http://security.debian.org unstable/updates/main all Packages
Err:12 http://security.debian.org unstable/updates/contrib armel Packages
  404  Not Found [IP: 2a04:4e42:1b::204 80]
Ign:13 http://security.debian.org unstable/updates/contrib all Packages
Ign:10 http://archive.ev3dev.org/debian unstable/main all Packages
Ign:15 http://security.debian.org unstable/updates/main armel Packages
Ign:16 http://security.debian.org unstable/updates/non-free all Packages
Ign:17 http://security.debian.org unstable/updates/non-free armel Packages
Err:14 http://archive.ev3dev.org/debian unstable/main armel Packages
  404  Not Found [IP: 2604:a880:2:d0::2089:e001 80]
Get:18 http://httpredir.debian.org/debian unstable/main armel Packages [7956 kB]
Get:20 http://httpredir.debian.org/debian unstable/non-free armel Packages [53.6 kB]
Fetched 8246 kB in 1min 57s (70.5 kB/s)
Reading package lists... 90%
```

Thank you for helping :)

dlech, 2020-02-09:
`apt` is really slow, how long did you wait? Depending on the speed of your microSD card it could take several minutes.

Also, that tutorial is probably out of date. ev3dev-stretch already has GCC 6, so using packages from unstable is not necessary.

Also consider using Docker to cross-compile if your project is large or uses libraries other than the standard library. https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

Niels1006, 2020-02-10:
Okay, imma try using Docker, thank you :)