ev3dev/ev3dev #1381: wifi wpa2 enterprise peap

snape6666, 2020-02-21:
Hi,
we are using your image and trying to connect with our enterprise wpa2 (peap) network.

How do we have to configure the wifi to get it running? We tried a lot of settings, but never get it running.
[image]

thx for your help in advance!

dlech, 2020-02-21:
ev3dev uses ConnMan for network management. So don't edit `/etc/wpa_supplicant` directly. Use the `connmanctl` command line tool instead. Arch Linux has a decent wiki page about it. https://wiki.archlinux.org/index.php/ConnMan

benderfzi, 2020-02-24:
Hi, we've read the ArchLinux wiki page you sent us and tried different configurations again. But all we get from connman on connect is:

connect wifi_74da38f79396_465a49_managed_ieee8021x
Error /net/connman/service/wifi_74da38f79396_465a49_managed_ieee8021x: Operation aborted

Where can we find debug logs? I get absolutely no clue from connman on what I'm doing wrong.

dlech, 2020-02-24:
Try `sudo journalctl -u connman.service` to get the logs.

benderfzi, 2020-02-24:
Hello David, thanks for your response. I was able to launch the connman service in debug mode and pulled the log as you said. I'm not posting the whole thing at the moment but rather two parts that at least contained some errors. Note that I rebooted the brick and then tried to connect to the wifi. This was the result:
`[...]
Mar 04 20:09:12 ev3dev connmand[455]: Adding service configuration wifi_74da38f79396_465a49_managed_ieee8021x
Mar 04 20:09:12 ev3dev connmand[455]: ../src/device.c:__connman_device_init()
Mar 04 20:09:12 ev3dev connmand[455]: ../src/ippool.c:__connman_ippool_init()
Mar 04 20:09:12 ev3dev connmand[455]: ../src/firewall-iptables.c:__connman_firewall_init()
Mar 04 20:09:12 ev3dev connmand[455]: ../src/iptables.c:__connman_iptables_init()
Mar 04 20:09:12 ev3dev connmand[455]: ../src/iptables.c:iptables_init() filter
Mar 04 20:09:13 ev3dev connmand[455]: ../src/iptables.c:iptables_init() iptable_filter module loading gives error but trying anyway
Mar 04 20:09:13 ev3dev connmand[455]: iptables support missing error 2 (No such file or directory)
Mar 04 20:09:13 ev3dev connmand[455]: ../src/iptables.c:__connman_iptables_commit() filter
Mar 04 20:09:13 ev3dev connmand[455]: Failed to flush table 'filter': Invalid argument
Mar 04 20:09:13 ev3dev connmand[455]: ../src/iptables.c:iptables_init() mangle
Mar 04 20:09:13 ev3dev connmand[455]: ../src/iptables.c:iptables_init() iptable_mangle module loading gives error but trying anyway
Mar 04 20:09:13 ev3dev connmand[455]: iptables support missing error 2 (No such file or directory)
Mar 04 20:09:13 ev3dev connmand[455]: ../src/iptables.c:__connman_iptables_commit() mangle
Mar 04 20:09:13 ev3dev connmand[455]: Failed to flush table 'mangle': Invalid argument
[...]
Mar 04 20:13:24 ev3dev connmand[455]: wlx74da38f79396 {RX} 7 packets 6693 bytes
Mar 04 20:13:24 ev3dev connmand[455]: wlx74da38f79396 {TX} 7 packets 605 bytes
Mar 04 20:13:24 ev3dev connmand[455]: wlx74da38f79396 {update} flags 36867 <UP>
Mar 04 20:13:24 ev3dev connmand[455]: wlx74da38f79396 {newlink} index 3 address 74:DA:38:F7:93:96 mtu 1500
Mar 04 20:13:24 ev3dev connmand[455]: wlx74da38f79396 {newlink} index 3 operstate 2 <DOWN>
Mar 04 20:13:24 ev3dev connmand[455]: ../src/detect.c:detect_newlink() type 1 index 3
Mar 04 20:13:24 ev3dev connmand[455]: ../plugins/wifi.c:wifi_newlink() index 3 flags 36867 change 0
Mar 04 20:13:24 ev3dev connmand[455]: ../plugins/wifi.c:wifi_newlink() carrier off
Mar 04 20:13:24 ev3dev connmand[455]: ../src/rtnl.c:rtnl_message() NEWLINK len 76 type 16 flags 0x0000 seq 0 pid 0
Mar 04 20:13:24 ev3dev connmand[455]: ../plugins/wifi.c:interface_state() wifi 0x5b5048 interface state 2
Mar 04 20:13:24 ev3dev connmand[455]: ../plugins/wifi.c:is_idle() state 7
Mar 04 20:13:24 ev3dev connmand[455]: ../src/network.c:connman_network_set_connected() network 0x5bbb20 connected 0/0 connecting 1 associating 1
Mar 04 20:13:24 ev3dev connmand[455]: ../src/network.c:connman_network_set_error() network 0x5bbb20 error 4
Mar 04 20:13:24 ev3dev connmand[455]: ../src/service.c:__connman_service_indicate_error() service 0x5bbc90 error 4
Mar 04 20:13:24 ev3dev connmand[455]: ../src/service.c:__connman_service_ipconfig_indicate_state() service 0x5bbc90 (wifi_74da38f79396_465a49_managed_ieee8021x) old state 2 (association) new state 7 (failure) type 1 (IPv4)
Mar 04 20:13:24 ev3dev connmand[455]: ../src/ipconfig.c:__connman_ipconfig_disable() ipconfig 0x5bbc48
Mar 04 20:13:24 ev3dev connmand[455]: ../src/service.c:service_indicate_state() service 0x5bbc90 old association - new failure/association => association
Mar 04 20:13:24 ev3dev connmand[455]: ../src/service.c:__connman_service_ipconfig_indicate_state() service 0x5bbc90 (wifi_74da38f79396_465a49_managed_ieee8021x) old state 2 (association) new state 7 (failure) type 2 (IPv6)
Mar 04 20:13:24 ev3dev connmand[455]: ../src/ipconfig.c:__connman_ipconfig_disable() ipconfig 0x5bbed8
Mar 04 20:13:24 ev3dev connmand[455]: ../src/service.c:service_indicate_state() service 0x5bbc90 old association - new failure/failure => failure
Mar 04 20:13:24 ev3dev connmand[455]: ../src/session.c:service_state_changed() service 0x5bbc90 state 7
[...]`

Can you spot anything unusual? Not sure what to do from here.

Thanks,

Janek

dlech, 2020-02-24:
There errors about iptables might mean that there are some Linux kernel modules that are not enabled, but I don't know if that would affect this particular type of network connection.

snape6666, 2020-02-24:
Try installing iptables (sudo apt-get install iptables).

snape6666, 2020-02-25:
still not working...

benderfzi, 2020-02-26:
It works now. We used our private networks at home to update the EV3 and then managed to connect to the company network. Thanks again.

closed 2020-02-27 by snape6666