ev3dev/ev3dev #1382: Device or resource busy

will493, 2020-02-21:
"Device or resource busy" error when trying to get access to /sys/class/pwm/pwmchip5 resource (ev3 + ev3dev-stretch img) after echo raw > /sys/class/lego-port/port4/mode

dlech, 2020-02-21:
EHRPWM1 is share by ports A and B so both ports probably need to be in raw mode or use port C or D.

will493, 2020-02-22:
robot@ev3dev:~$ echo raw > /sys/class/lego-port/port4/mode
robot@ev3dev:~$ echo raw > /sys/class/lego-port/port5/mode
robot@ev3dev:~$ echo 1 > /sys/class/pwm/pwmchip5/export
-bash: echo: write error: Device or resource busy
robot@ev3dev:~$ echo raw > /sys/class/lego-port/port6/mode
robot@ev3dev:~$ echo 0 > /sys/class/pwm/pwmchip0/export
-bash: echo: write error: Device or resource busy