ev3dev/ev3dev #1383: reading EV3 ColorSensor REF-RAW values in PyBricks

BrBarry, 2020-02-22:
**Question**

There is way not enough documentation around on PyBricks EV3 microPython....

I'm tring to improve my LineFollowing Code. I've seen others use complicated extraction of Analogue data with the combination of ev3dev2.port LegoPort and Pybricks.ev3devio Ev3devSensor.
However, I do see on the brick a REF-RAW mode on a EV3 Color Sensor.

How do I access and read the RAW values on 2 ports (Port.S2 & S3) and yes I know this might not the correct method of assessing in the ports.

JakubVanek, 2020-02-23:
If you are using the official EV3 color sensor, then it is not possible to read them using analog pins - the data are communicated via a serial line from the sensor. However, if you can get the `value0` and `value1` of the `REF-RAW` mode, then indeed these should be the raw data read by the sensor.

I think I have discovered the meaning behind these values, see https://github.com/simonedegiacomi/EV3-API/commit/20a57f7b944a1e53477ebdfe61f96ebf2d6c66b7#r37442226 .

However I don't know how to access the values in PyBricks.

dlech, 2020-02-23:
You can use `color_sensor._mode('RGB-RAW')` to set the mode and then use

    red = color_sensor._value(0)
    green = color_sensor._value(1)
    blue = color_sensor._value(2)

to read the raw values.

BrBarry, 2020-02-24:
Mode set correctly, However I getting AttributeError when trying to read it.
can I get the attribute for REF-RAW to read the values please.

dlech, 2020-02-24:
If you are getting the error "'AttributeError': 'ColorSensor' object has no attribute 'Raw'", then I'm guessing that you wrote `ColorSensor.Raw` instead of or in addition to what I suggested.

BrBarry, 2020-02-24:
As I drill down through various ports (look for answers to many other questions). I tried @LaiYanKai suggestion noted in Issue #1292. Where he suggested using the "help" via SSH terminal to get hints on methods available. 
Although the mode on the brick allow for REF-RAW and code changing the mode works. When querying ColorSensor using help.ColorSensor (abbreviated here), there is no such function or method relating to REF-RAW in ColorSensor. Only the other 3.

So I guess, at this point in time a cannot access the raw reflection data from the colour sensor. Just that I read in here somewhere else that it could have a better resolution and a wider data range to work with.
Using it in my own PID code of a drivebase with two colour sensors.

dlech, 2020-02-25:
The `_value()` method works for all modes.

Here is a full working example:

```python
#!/usr/bin/env pybricks-micropython

from pybricks import ev3brick as brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import (Port, Stop, Direction, Button, Color,
                                 SoundFile, ImageFile, Align)
from pybricks.tools import print, wait, StopWatch
from pybricks.robotics import DriveBase

# Write your program here
brick.sound.beep()

color_sensor = ColorSensor(Port.S1)
color_sensor._mode('REF-RAW')

while True:
    raw1, raw2 = color_sensor._value(0), color_sensor._value(1)
    print(raw1, raw2)
    wait(1000)
```

JakubVanek, 2020-02-25:
Extending the example by @dlech to include processing of the raw values:
```python3
#!/usr/bin/env pybricks-micropython

from pybricks import ev3brick as brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import (Port, Stop, Direction, Button, Color,
                                 SoundFile, ImageFile, Align)
from pybricks.tools import print, wait, StopWatch
from pybricks.robotics import DriveBase

# limit value to a bounded range
def clamp(x, min_x, max_x):
    if x < min_x:
        return min_x
    elif x > max_x:
        return max_x
    else
        return x

# calibrated max delta value for a given surface
max_delta = 400

brick.sound.beep()

# initialize sensor
color_sensor = ColorSensor(Port.S1)
color_sensor._mode('REF-RAW')

while True:
    # fetch raw ADC reading from inside the sensor
    raw_reflection, raw_ambient = color_sensor._value(0), color_sensor._value(1)

    # calculate the difference between measurements when the LED is on and when it's off
    # the ordering should be correct - higher photosensor illumination causes lower voltage to be measured
    delta = raw_ambient - raw_reflection

    # normalize reading to range 0.0~1.0
    fractional = delta / max_delta
    bounded = clamp(fractional, 0.0, 1.0)

    # print data out
    print("--------")
    print("Raw reflection ADC value : {:4d}"       .format(raw_reflection))
    print("Raw ambient    ADC value : {:4d}"       .format(raw_ambient))
    print("Reading difference       : {:4d} / {:d}".format(delta, max_delta))
    print("Normalized reflectance   : {:7.3f}%"    .format(bounded * 100.0))
```

BrBarry, 2020-02-26:
Thanks to both for your help.
Especially the expanded example. A good scope of data to test with.

I'll Close of with these answers

closed 2020-02-26 by BrBarry

DanieleBenedettelli, 2023-02-20:
I need to revive this old thread I found while looking for a solution to the same problem.
The _mode method does not exist in the newest API. 
To access raw the Color Sensor RGB values, according to https://pybricks.com/ev3-micropython/iodevices.html,
now you need to write:
`from pybricks.iodevices import Ev3devSensor    
from pybricks.parameters import Port                             
sensorBlocksRaw = Ev3devSensor(Port.S4)

(r,g,b) = sensorBlocksRaw.read('RGB-RAW')

`