ev3dev/ev3dev #1387: Is there any information about the "secret" EV3 gyro commands?

JakubVanek, 2020-02-23:
**Question**
Are there more known information regarding the pseudo-commands that can be sent to the EV3 gyro?

**Background Information**

This is driven by my curiosity, (hopefully) no XY problem here.

I was recently surprised that there exists a `INPUT_WRITE` opcode/`direct` attribute for writing data directly to the LEGO UART sensors. I tried grepping through the stock firmware and I found interesting stuff:

```
c_input/source/c_input.c
389:IMGDATA CLR_LAYER_INPUT_WRITE[]       = { opINPUT_WRITE,0,0,1,DEVCMD_RESET,opINPUT_WRITE,0,1,1,DEVCMD_RESET,opINPUT_WRITE,0,2,1,DEVCMD_RESET,opINPUT_WRITE,0,3,1,DEVCMD_RESET,opOBJECT_END };
398:  ExecuteByteCode(CLR_LAYER_INPUT_WRITE,NULL,NULL);
5016: *  <b>     opINPUT_WRITE (LAYER, NO, BYTES, DATA)  </b>
5026:/*! \brief  opINPUT_WRITE byte code

lms2012/doc/html/cInput.html
863:<p> <b> opINPUT_WRITE (LAYER, NO, BYTES, DATA) </b></p>

lms2012/doc/html/testuslevelappcode.html
151:  INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-1')                          //        INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-0')
158:  INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-2')                          //        INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-1')
165:  INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-3')                          //        INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-2')

lms2012/doc/html/brickprogramtemplateapp.html
1232:  INPUT_WRITE(0,GYRO_PORT,1,DEVCMD_RESET)                             //    INPUT_WRITE(0,GYRO_PORT,1,DEVCMD_RESET)

lms2012/doc/html/testdetectcalappcode.html
161:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-1')                      //
169:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-2')                      //

lms2012/doc/html/applicationversions.html
1064:<li>opINPUT_WRITE make device not ready until new data arrive <br />
1376:<li>opINPUT_WRITE implemented (only works on UART device) <br />

lms2012/doc/html/testtiltcalappcode.html
155:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-1')                        //
163:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-2')                        //
317:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-1')                        //
325:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-2')                        //

lms2012/doc/html/bytecodedef.html
256:  opINPUT_WRITE               = 0x9F, //       111

lms2012/doc/html/test.html
1870:  INPUT_WRITE(0,GYRO_PORT,14,'LEGO-FAC-CAL-1')                        //
1880:  INPUT_WRITE(0,GYRO_PORT,14,'LEGO-FAC-CAL-2')                        //
2015:  INPUT_WRITE(0,COLOR_PORT,14,'LEGO-FAC-CAL-1')                        //
2147:  INPUT_WRITE(0,SEEK_PORT,14,'LEGO-FAC-CAL-1')                        //
2396:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-1')                     //
2402:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-1')                     //
2438:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-2')                     //
2444:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-2')                     //

lms2012/doc/Implementation
664: *-# opINPUT_WRITE make device not ready until new data arrive                                \n
842: *-#  opINPUT_WRITE implemented (only works on UART device)                                   \n

lms2012/source/bytecodes.c
315:  OC(   opINPUT_WRITE,          PAR8,PAR8,PAR8,PAR8,                            0,0,0,0               ),

lms2012/source/bytecodes.h
404:  opINPUT_WRITE               = 0x9F, //       111

lms2012/source/lms2012.c
2828:  [opINPUT_WRITE]         =   &cInputWrite,

lmssrc/adk/lmststasm/bytecodes.h
404:  opINPUT_WRITE               = 0x9F, //       111

lmssrc/adk/lmsasm/bytecodes.c
315:  OC(   opINPUT_WRITE,          PAR8,PAR8,PAR8,PAR8,                            0,0,0,0               ),

lmssrc/adk/lmsasm/DebugBytecodePrograms/GyroAutoZero.lms
29:  INPUT_WRITE(0,GYROPORT,1,0x88)

lmssrc/adk/lmsasm/bytecodes.h
404:  opINPUT_WRITE               = 0x9F, //       111

lmssrc/adk/lmsasm/DebugBytecodePrograms/GyroNoAutoZero.lms
29:  INPUT_WRITE(0,GYROPORT,1,0x11)

lmssrc/adk/lmsasm/DebugBytecodePrograms/Bytecode.lms
154:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-1')                        //
162:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-2')                        //

lmssrc/adk/lmsasm/DebugBytecodePrograms/Bytecode.lms~
154:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-1')                      //
162:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-2')                      //

lmssrc/Test/Test.lms
1239:  INPUT_WRITE(0,0,2,Data)                        
1265:  INPUT_WRITE(0,0,2,Data)                        
1280:  INPUT_WRITE(0,0,2,Data)                        
1295:  INPUT_WRITE(0,0,2,Data)                        
1334:  INPUT_WRITE(0,0,2,Data)                        
1359:  INPUT_WRITE(0,0,2,Data)                        
1374:  INPUT_WRITE(0,0,2,Data)                        
1389:  INPUT_WRITE(0,0,2,Data)                        
1428:  INPUT_WRITE(0,0,2,Data)                        
1453:  INPUT_WRITE(0,0,2,Data)                        
1468:  INPUT_WRITE(0,0,2,Data)                        
1483:  INPUT_WRITE(0,0,2,Data)                        
1847:  INPUT_WRITE(0,GYRO_PORT,14,'LEGO-FAC-CAL-1')                        //
1857:  INPUT_WRITE(0,GYRO_PORT,14,'LEGO-FAC-CAL-2')                        //
1992:  INPUT_WRITE(0,COLOR_PORT,14,'LEGO-FAC-CAL-1')                        //
2124:  INPUT_WRITE(0,SEEK_PORT,14,'LEGO-FAC-CAL-1')                        //
2373:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-1')                     //
2379:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-1')                     //
2415:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-2')                     //
2421:  INPUT_WRITE(0,TT_GYRO_PORT,14,'LEGO-FAC-CAL-2')                     //
2583:  INPUT_WRITE(0,US_PORT,1,DEVCMD_FIRE)

lmssrc/TEST/Tilt Cal.lms
155:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-1')                        //
163:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-2')                        //

lmssrc/TEST/Detect Cal.lms
160:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-1')                      //
168:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-2')                      //

lmssrc/TEST/Detect Cal.lms~
160:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-1')                      //
168:  INPUT_WRITE(0,DETECT_PORT,14,'LEGO-FAC-CAL-2')                      //

lmssrc/TEST/USlevel.lms
150:  INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-1')                          //        INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-0')
157:  INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-2')                          //        INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-1')
164:  INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-3')                          //        INPUT_WRITE(0,US_PORT,14,'LEGO-FAC-CAL-2')

lmssrc/TEST/Tilt Cal.lms~
155:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-1')                        //
163:  INPUT_WRITE(0,TILT_PORT,14,'LEGO-FAC-CAL-2')                        //

lmssrc/tst/us.lms
33:  INPUT_WRITE(0,US_PORT,1,17)

lmssrc/Brick Program/template.lms
1227:  INPUT_WRITE(0,GYRO_PORT,1,DEVCMD_RESET)                             //    INPUT_WRITE(0,GYRO_PORT,1,DEVCMD_RESET)
```

Currently I'm focusing on the gyro sensor. From the output above, we can deduce that the sensor supports the following commands:
* `DEVCMD_RESET` = 0x11 = 17 in the `GYRO-ANGLE` mode. I have found this first in https://github.com/ev3dev/ev3dev-lang-python/issues/362 and https://github.com/ev3dev/ev3dev/issues/1372. According to these sources, this resets the angle value and calibrates the gyro sensor.
* `"LEGO-FAC-CAL-1"` in the `GYRO-CAL` mode. This appears to start some sort of [calibration 1](https://github.com/mindboards/ev3sources/blob/78ebaf5b6f8fe31cc17aa5dce0f8e4916a4fc072/lms2012/lmssrc/Test/Test.lms#L1847).
* `"LEGO-FAC-CAL-2"` in the `GYRO-CAL` mode. This appears to start some sort of [calibration 2](https://github.com/mindboards/ev3sources/blob/78ebaf5b6f8fe31cc17aa5dce0f8e4916a4fc072/lms2012/lmssrc/Test/Test.lms#L1857).
* `0x88`. This is the command that made me create this issue. It can be found in a `GyroAutoZero.lms` file and hints towards that there is some automatic calibration present in the sensor. I have no idea if it works actually, but it looks intriguing.

**Robot (please complete the following information):**
 - Device: EV3
 - Gyro sensor revision (not important though, this question is meant in general): 49N3

JakubVanek, 2020-02-23:
I have done an experiment with hacked @simonedegiacomi's c4ev3 sensors branch and I have got interesting results. I tried sending commands while continuously reading sensor data: 
* `GYRO-ANG`
  * sending `0x11` (GyroNoAutoZero) resets the angle offset, *but* it does not calibrate the sensor.
  * sending `0x88` (GyroAutoZero) resets the angle offset *and* calibrates the gyro sensor.
  * switching to `GYRO-CAL` and back resets the angle offset and somethimes calibrates the sensor, but sometimes it appears like it just sets some random value.
* `GYRO-RATE`
  * sending `0x11` (GyroNoAutoZero) calibrates the sensor
  * sending `0x88` (GyroAutoZero) calibrates the sensor
  * switching to `GYRO-CAL` and back calibrates the sensor
* `GYRO-FAS`
  * sending `0x11` (GyroNoAutoZero) sometimes calibrates the sensor
  * sending `0x88` (GyroAutoZero) calibrates the sensor
  * switching to `GYRO-CAL` and back calibrates the sensor
* `GYRO-GnA`
  * sending `0x11` (GyroNoAutoZero) has no effect (known bug in this sensor revision)
  * sending `0x88` (GyroAutoZero) has no effect (known bug in this sensor revision)
  * switching to `GYRO-CAL` and back has no effect (known bug in this sensor revision)

I have no idea if the gyro recalibration is caused by the sensor timing out or if the sensor really interprets the command and performs intentional recalibration.

JakubVanek, 2020-02-23:
Sending `0x08` or `0x80` or `0x44` seems to have no effect, therefore `0x88` has to have some special meaning.

JakubVanek, 2020-02-23:
Source code of the hacked library & test application (test/test.c): https://github.com/JakubVanek/EV3-API/tree/ugly-gyro-hack

JakubVanek, 2020-02-23:
I have tried sending all single-byte commands to the sensor while in `GYRO-ANGLE` mode. The only ones that caused the measured angle to change were the 0x11 and 0x88 commands. However this does not rule out existence of more commands, perhaps behaving like I2C register writes.

dlech, 2020-02-23:
On gyro sensors that have a date code that ends with N2 or N3, changing mode causes the sensor to reboot, which has the side-effect of "calibrating" the sensor. However newer versions don't reboot on mode switch, so it would be interesting to repeat these tests on newer sensors.

dlech, 2020-02-23:
> lmssrc/adk/lmsasm/DebugBytecodePrograms/GyroAutoZero.lms
> 29:  INPUT_WRITE(0,GYROPORT,1,0x88)

Where did you find this?

JakubVanek, 2020-02-23:
> > lmssrc/adk/lmsasm/DebugBytecodePrograms/GyroAutoZero.lms
> > 29:  INPUT_WRITE(0,GYROPORT,1,0x88)
> 
> Where did you find this?

I think I have downloaded this source code revision directly from LEGO (it contains the 1.09 version): https://education.lego.com/en-us/support/mindstorms-ev3/developer-kits → https://le-www-live-s.legocdn.com/sc/media/files/ev3-developer-kit/lego-mindstorms-ev3-firmware-source-code-d7a19f70086183fc82354f8de659a2da.bz2. I have just now noticed that the mindboards/ev3sources does not have these files.

JakubVanek, 2020-02-24:
> On gyro sensors that have a date code that ends with N2 or N3, changing mode causes the sensor to reboot, which has the side-effect of "calibrating" the sensor. However newer versions don't reboot on mode switch, so it would be interesting to repeat these tests on newer sensors.

I don't have access to the newer sensors, but I have ported the test program to python-ev3dev2:
```python3
#!/usr/bin/env python3

from time import sleep

from ev3dev2.button import Button
from ev3dev2.sensor.lego import GyroSensor

def main():
    print("EV3 Gyro reset tester")
    print("Controls:")
    print(" - down:      do partial calibration (0x11)")
    print(" - up:        do full    calibration (0x88)")
    print(" - right:     do a switch to COL-CAL and back")
    print(" - backspace: go to next sensor mode")

    gyro = GyroSensor()
    angle_loop(gyro)
    rate_loop(gyro)
    fas_loop(gyro)
    gna_loop(gyro)
    tilt_angle_loop(gyro)
    tilt_rate_loop(gyro)

def do_full_calibration(gyro):
    gyro._direct = gyro.set_attr_raw(gyro._direct, 'direct', b'\x88')

def do_partial_calibration(gyro):
    gyro._direct = gyro.set_attr_raw(gyro._direct, 'direct', b'\x11')

def handle_keys(gyro):
    if Button.down:
        do_partial_calibration(gyro)
    if Button.up:
        do_full_calibration(gyro)
    if Button.right:
        old_mode = gyro.mode
        gyro.mode = "GYRO-CAL"
        gyro.mode = old_mode

def angle_loop(gyro):
    print("switching to GYRO-ANG mode")
    while not Button.backspace:
        print(f"angle: {gyro.angle}°")
        handle_keys(gyro)
        sleep(0.1)

    print("leaving GYRO-ANG mode")
    sleep(1)

def rate_loop(gyro):
    print("switching to GYRO-RATE mode")
    while not Button.backspace:
        print(f"rate: {gyro.rate}°/s")
        handle_keys(gyro)
        sleep(0.1)

    print("leaving GYRO-RATE mode")
    sleep(1)

def fas_loop(gyro):
    print("switching to GYRO-FAS mode")
    while not Button.backspace:
        gyro._ensure_mode(gyro.MODE_GYRO_FAS)
        print(f"rate: {gyro.value(0)}°/s")
        handle_keys(gyro)
        sleep(0.1)

    print("leaving GYRO-RATE mode")
    sleep(1)

def gna_loop(gyro):
    print("switching to GYRO-G&A mode (final step for N2/N3 sensors)")
    while not Button.backspace:
        angle, rate = gyro.angle_and_rate
        print(f"angle: {angle}°, rate: {rate}°/s")
        handle_keys(gyro)
        sleep(0.1)

    print("leaving GYRO-G&A mode")
    sleep(1)

def tilt_angle_loop(gyro):
    print("switching to TILT-ANG mode")
    while not Button.backspace:
        print(f"angle: {gyro.tilt_angle}°")
        handle_keys(gyro)
        sleep(0.1)

    print("leaving TILT-ANG mode")
    sleep(1)

def tilt_rate_loop(gyro):
    print("switching to TILT-RATE mode")
    while not Button.backspace:
        print(f"rate: {gyro.tilt_rate}°/s")
        handle_keys(gyro)
        sleep(0.1)

    print("leaving TILT-RATE mode")
    sleep(1)

if __name__ == "__main__":
    main()
```

JakubVanek, 2020-03-09:
Hmmm, I'm now experimenting with this again and for some reason, the commands do not work for the `GYRO-RATE` mode (on the same N3 gyro sensor). I don't know if this is due to my error or some bug in the previous code, but now `0x11` doesn't do anything and `0x88` freezes the values (only a modeswitch helps).

dlech, 2020-03-09:
As far as I know 0x11 only works on the GYRO-ANGLE mode (EV3-G gyro reset block switches to this mode before sending this command). And I've not seen 0x88 ever used anywhere so it is hard to say if it was actually intended to be used or tested much during development.

JakubVanek, 2020-03-10:
Oh, I've found it...
```c
int HackEV3GyroSensorCmd(int port, uint8_t cmd) {
    // [comment] this was the bad idea
    // I haven't realized that this WILL perform an extra modeswitch and so it will reboot my N3 sensor
    // therefore the tables in my previous comments are mostly wrong
    setEV3GyroSensorMode(port, EV3_GYRO_SENSOR_ANGLE_MODE);

    uint8_t buffer[2];
    buffer[0] = port;
    buffer[1] = cmd;

    return writeToUART(port, buffer, sizeof(buffer));
}
```

Yup, the new tests confirm that 0x11 works only in GYRO-ANGLE. Luckily at least 0x88 seems to still work when used in GYRO-ANGLE. However the commands may still be useful on the N4+ sensors.

I would like to ask, would you please test the commands for me on some newer sensors?
```python
#!/usr/bin/env python3

from time import sleep

from ev3dev2.button import Button
from ev3dev2.sensor.lego import GyroSensor

def main():
    print("EV3 Gyro reset tester")
    print("Usage:")
    print("  ./gyrotest.py [sensor mode name]")
    print("Controls:")
    print(" - down:      do partial calibration (0x11)")
    print(" - up:        do full    calibration (0x88)")
    print(" - right:     do a few mode switches")
    print(" - backspace: exit")
    
    mode = sys.argv[1]

    gyro = GyroSensor()

    while not Button.backspace:
        if mode == 'GYRO-ANG':
            print(f"angle: {gyro.angle:6d}°")

        elif mode == 'GYRO-RATE':
            print(f"rate:  {gyro.rate:4d}°/s")

        elif mode == 'GYRO-GnA':
            angle, rate = gyro.angle_and_rate
            print(f"angle: {gyro.angle:6d}°, rate: {gyro.rate:4d}°/s")

        elif mode == 'GYRO-FAS':
            gyro._ensure_mode(gyro.MODE_GYRO_FAS)
            print(f"fast rate: {gyro.value(0)}°/s")

        elif mode == 'TILT-ANG':
            print(f"tilt angle: {gyro.tilt_angle:6d}°")

        elif mode == 'TILT-RATE':
            print(f"tilt rate:  {gyro.tilt_rate:4d}°/s")

        if Button.up:
            do_full_calibration()
            while Button.up:
                sleep(0.1)

        if Button.down:
            do_partial_calibration()
            while Button.down:
                sleep(0.1)

        if Button.left:
            do_mode_switches()
            while Button.left:
                sleep(0.1)

        sleep(0.1)

def do_full_calibration(gyro):
    print("SENDING COMMAND: FULL CALIBRATION")
    old_mode = gyro.mode
    gyro._ensure_mode(gyro.MODE_GYRO_ANG)
    gyro._direct = gyro.set_attr_raw(gyro._direct, 'direct', b'\x88')
    gyro._ensure_mode(old_mode)

def do_partial_calibration(gyro):
    print("SENDING COMMAND: OFFSET RESET")
    old_mode = gyro.mode
    gyro._ensure_mode(gyro.MODE_GYRO_ANG)
    gyro._direct = gyro.set_attr_raw(gyro._direct, 'direct', b'\x11')
    gyro._ensure_mode(old_mode)

def do_mode_switches(gyro):
    print("SWITCHING MODES")
    old_mode = gyro.mode
    gyro.mode = 'GYRO-ANG'
    gyro.mode = 'GYRO-RATE'
    gyro.mode = old_mode

if __name__ == "__main__":
    main()
```
I think the most interesting modes for testing could be `GYRO-G&A` and `TILT-ANGLE`:
 - Does the partial calibration command reset the angle part of G&A?
 - Does the full calibration command remove the bias from both G&A values?
 - Does the partial calibration command reset the TILT-ANGLE value to zero?
 - Does the full calibration command change the drift in TILT-ANGLE?

JakubVanek, 2020-03-10:
If the tests reveals that `0x88` works on new sensors too, it means that there is a universal calibration sequence:

>  1. switch to `GYRO-ANGLE`
>  2. send the full calibration command
>  3. switch back to old mode
> 
> 
> This should work both on N2/N3 sensors (where modeswitch and commands work) and it could work on N4+ too (where only commands have a chance of working).

dlech, 2020-03-11:
I tried this on an N3, an N4 and an N5 sensor. Unfortunately, the 0x88 command does not work on the newer sensors. It just acts the same as 0x11 as far as I can tell. Also (not 100% sure about this), it seems like the first time I tried the N3 sensor 0x88 didn't do the full calibration either but it worked on the second attempt after trying the other sensor versions.

dlech, 2020-03-11:
And in case anyone else wants to try, here is a working program (had to make some minor changes to get it to run).

```python
#!/usr/bin/env python3

import sys
from time import sleep

from ev3dev2.button import Button
from ev3dev2.sensor.lego import GyroSensor

def main():
    print("EV3 Gyro reset tester")
    print("Usage:")
    print("  brickrun -r -- ./gyrotest.py [sensor mode name]")
    print("Controls:")
    print(" - down:      do partial calibration (0x11)")
    print(" - up:        do full    calibration (0x88)")
    print(" - left:      do a few mode switches")

    mode = sys.argv[1]

    gyro = GyroSensor()
    btn = Button()

    while True:
        if mode == 'GYRO-ANG':
            print("angle: {:6d}deg".format(gyro.angle))

        elif mode == 'GYRO-RATE':
            print("rate:  {:4d}deg/s".format(gyro.rate))

        elif mode == 'GYRO-GnA':
            angle, rate = gyro.angle_and_rate
            print("angle: {:6d}deg, rate: {:4d}deg/s".format(
                gyro.angle, gyro.rate))

        elif mode == 'GYRO-FAS':
            gyro._ensure_mode(gyro.MODE_GYRO_FAS)
            print("fast rate: {}deg/s".format(gyro.value(0)))

        elif mode == 'TILT-ANG':
            print("tilt angle: {:6d}deg".format(gyro.tilt_angle))

        elif mode == 'TILT-RATE':
            print("tilt rate:  {:4d}deg/s".format(gyro.tilt_rate))

        if btn.up:
            do_full_calibration(gyro)
            while btn.up:
                sleep(0.1)

        if btn.down:
            do_partial_calibration(gyro)
            while btn.down:
                sleep(0.1)

        if btn.left:
            do_mode_switches(gyro)
            while btn.left:
                sleep(0.1)

        sleep(0.1)

def do_full_calibration(gyro):
    print("SENDING COMMAND: FULL CALIBRATION")
    old_mode = gyro.mode
    gyro._ensure_mode(gyro.MODE_GYRO_ANG)
    gyro._direct = gyro.set_attr_raw(gyro._direct, 'direct', b'\x88')
    gyro._ensure_mode(old_mode)

def do_partial_calibration(gyro):
    print("SENDING COMMAND: OFFSET RESET")
    old_mode = gyro.mode
    gyro._ensure_mode(gyro.MODE_GYRO_ANG)
    gyro._direct = gyro.set_attr_raw(gyro._direct, 'direct', b'\x11')
    gyro._ensure_mode(old_mode)

def do_mode_switches(gyro):
    print("SWITCHING MODES")
    old_mode = gyro.mode
    gyro.mode = 'GYRO-ANG'
    gyro.mode = 'GYRO-RATE'
    gyro.mode = old_mode

if __name__ == "__main__":
    main()
```

JakubVanek, 2020-03-11:
Oh, that is a bit disappointing, however thank you for the testing!