ev3dev/ev3dev #1389: Bluetooth not avaliable

bemxio, 2020-02-28:
**What are you trying to do?**

I am trying to turn on Bluetooth

**What did you expect to happen**

Just turning on and the configuration will be avaliable

**What actually happened**

it just says "not avaliable", even when i restart, it's the same

**Screenshots**
none

**Desktop (please complete the following information):**
 - OS and version: [Windows 10 1903]
 - Connection Type: [Bluetooth]

**Robot (please complete the following information):**
 - Device: [EV3]
 - Kernel: 4.14.117-ev3dev-2.3.4-ev3
 - Model: LEGO MINDSTORMS EV3
 - Revision: 0000

dlech, 2020-02-28:
See #1314

bemxio, 2020-02-29:
nvm, i decided to use usb, but i have another problem (issue #1390)

closed 2020-02-29 by bemxio