ev3dev/ev3dev #1390: ConnectionRefusedError: [WinError 10061]

bemxio, 2020-02-29:
**What are you trying to do?**

I am trying to run the script in the shell

**What did you expect to happen**

The program should have play sound

**What actually happened**

The program did traceback and the error i mentioned in title

**Source Code**

Here is the program I am trying to run.

```
#!/usr/bin/env python3
from ev3dev2.sound import Sound

sound = Sound()
sound.speak('Welcome to the E V 3 dev project!')
```

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS and version: Windows 10 1903
 - Connection Type: USB
 - Software: Thonny, PuTTY

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```

Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=0016537F7FF1
BOARD_INFO_TYPE=main

```

**Background Information**
I am trying to make a script that runs in shell and communicates from the shell to EV3 because I want to build remote control for my EV3 robotic hand

dlech, 2020-02-29:
Can you give the full steps to reproduce the problem?

bemxio, 2020-02-29:
I don't know how to do it exactly but I will say what I did:
1. plugging ev3 to pc by mini usb
2. doing everything from the tutorial to connect ev3 to internet via usb
3. installing thonny (ide for python) ev3 extention
4. writing this code

dlech, 2020-02-29:
I don't know anything about thonny. Maybe ask for help at https://github.com/ev3dev-python-tools/thonny-ev3dev?

bemxio, 2020-02-29:
ok, I will ask them

closed 2021-03-08 by bemxio