ev3dev/ev3dev #1391: Brickpi3 vscode not working

Cptmeme, 2020-03-06:
**What are you trying to do?**

I am trying to learn how to code my brickpi using vscode (python)

**What did you expect to happen**

when i run this program on my ev3 brick with ev3dev on it, it does what it is supposed to do (5 rotations) so I would expect it to do the same on my brickpi3

**What actually happened**

It says its running but the motor doesnt do anything

**Code:**

#!/usr/bin/env python3
from ev3dev2.motor import LargeMotor,OUTPUT_A, SpeedPercent

m = LargeMotor(OUTPUT_A)
m.on_for_rotations(SpeedPercent(75), 5)

 - OS and version: Windows 10 
 - Connection Type: Wi-Fi
 - Software: Visual Studio Code

Image file:         ev3dev-stretch-rpi2-generic-2019-10-23
Kernel version:     4.14.114-ev3dev-2.3.4-rpi2
Brickman:           0.10.2
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000008468779b
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.8
BOARD_INFO_HW_REV=3.0.0
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=6F08C0F1514D355934202020FF0E183E
BOARD_INFO_TYPE=aux

Cptmeme, 2020-03-07:
After some testing it turns out that Output A and C do not work, but B and D do work as exspected. Are there codelines I have to edit (I.e. in Brickpi3.py)? Although the code seems to be correct according to "www.ev3dev.org/docs/platform-comparison/". 

Thanks

Cptmeme, 2020-03-18:
does anyone know why it doesnt work?

dlech, 2020-03-18:
Do ports A and C work with Dexter Industries' software?

Cptmeme, 2020-03-18:
yes they work with the official software
I also found out that all the sensor ports dont work with ev3dev (but do work with the official software).
thanks

dlech, 2020-03-18:
Can you share your code?

Cptmeme, 2020-03-18:
What code do you mean? The program i am trying to run is in my first comment.

dlech, 2020-03-18:
I do not see any code for sensors.

Cptmeme, 2020-03-18:
#!/usr/bin/env python3
from ev3dev2.motor import LargeMotor, OUTPUT_A, OUTPUT_D, SpeedPercent, MoveTank, MoveSteering
from ev3dev2.sensor import INPUT_4
from ev3dev2.sensor.lego import InfraredSensor
import time
Steering = MoveSteering(OUTPUT_D, OUTPUT_A)
s = InfraredSensor(INPUT_4)
s.heading(channel=1)
m = MoveTank(OUTPUT_D, OUTPUT_A)
s.distance(channel=1)

while True:
    if s.heading(channel=1) != 0:
        if s.heading(channel=1) < 0:
            Steering.on(100, 10)
        elif s.heading(channel=1) > 0:
            Steering.on(-100, 10)
    else:
        m.on_for_seconds(50,50,1)
    time.sleep(0.1)

Cptmeme, 2020-03-18:
`#!/usr/bin/env python3
from ev3dev2.motor import LargeMotor, OUTPUT_A, OUTPUT_D, SpeedPercent, MoveTank, MoveSteering
from ev3dev2.sensor import INPUT_4
from ev3dev2.sensor.lego import InfraredSensor
import time
Steering = MoveSteering(OUTPUT_D, OUTPUT_A)
s = InfraredSensor(INPUT_4)
s.heading(channel=1)
m = MoveTank(OUTPUT_D, OUTPUT_A)
s.distance(channel=1)

while True:
    if s.heading(channel=1) != 0:
        if s.heading(channel=1) < 0:
            Steering.on(100, 10)
        elif s.heading(channel=1) > 0:
            Steering.on(-100, 10)
    else:
        m.on_for_seconds(50,50,1)
    time.sleep(0.1)

`

Cptmeme, 2020-03-18:
``#!/usr/bin/env python3
from ev3dev2.motor import LargeMotor, OUTPUT_A, OUTPUT_D, SpeedPercent, MoveTank, MoveSteering
from ev3dev2.sensor import INPUT_4
from ev3dev2.sensor.lego import InfraredSensor
import time
Steering = MoveSteering(OUTPUT_D, OUTPUT_A)
s = InfraredSensor(INPUT_4)
s.heading(channel=1)
m = MoveTank(OUTPUT_D, OUTPUT_A)
s.distance(channel=1)

while True:
    if s.heading(channel=1) != 0:
        if s.heading(channel=1) < 0:
            Steering.on(100, 10)
        elif s.heading(channel=1) > 0:
            Steering.on(-100, 10)
    else:
        m.on_for_seconds(50,50,1)
    time.sleep(0.1)

``

dlech, 2020-03-18:
BrickPi3 cannot automatically detect sensors. You have to use the `LegoPort` object to configure the port and select the sensor type. Maybe something like this...

```python

p4 = LegoPort(INPUT_4)
p4.mode = 'ev3-uart'
p4.set_device = 'lego-ev3-ir'
sleep(0.5)

s = InfraredSensor(INPUT_4)
```

Cptmeme, 2020-03-18:
It doesnt give me error messages anymore but the beacon heading still doesnt work

WasabiFan, 2020-03-18:
@dlech Is `ev3-uart` definitely the right string here? I don't see it listed on this page: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/ports.html#list-of-port-drivers

dlech, 2020-03-18:
Comes from http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/brickpi3.html#brickpi3-in-port-modes

dlech, 2020-03-19:
```python
#!/usr/bin/env python3

"""
The Brickpi3 doesn't support auto-detecting motors and sensors. To use devices
connected to the input ports, you must specify what type of device it is.
Output ports are pre-configured as NXT Large motors and do not need to be
configured manually.
"""

from time import sleep
from ev3dev2 import list_devices
from ev3dev2.port import LegoPort
from ev3dev2.motor import OUTPUT_A, LargeMotor, SpeedPercent
from ev3dev2.sensor import INPUT_1
from ev3dev2.sensor.lego import UltrasonicSensor

p1 = LegoPort(INPUT_1)
# http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/brickpi3.html#brickpi3-in-port-modes
p1.mode = 'ev3-uart'
# http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html#supported-sensors
p1.set_device = 'lego-ev3-us'

# allow for some time to load the new drivers
sleep(0.5)

s = UltrasonicSensor(INPUT_1)
m = LargeMotor(OUTPUT_A)

print("Running motor...")

while True:
    dist = s.distance_centimeters
    if dist < 50:
        m.on(SpeedPercent(30))
    else:
        m.on(SpeedPercent(-30))

    sleep(0.05)
```

This program works for me.

```
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-rpi2-generic-2020-03-07
Kernel version:     4.14.114-ev3dev-2.3.5-rpi2
Brickman:           0.10.3
BogoMIPS:           38.40
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=A21041
BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
BOARD_INFO_SERIAL_NUM=000000004ed63807
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=FAD3CABB504B5354392E314BFF0F2C38
BOARD_INFO_TYPE=aux
```

dlech, 2020-03-19:
Maybe the difference is the BrickPi3 firmware version?

Running `sudo update-brickpi3-fw` on ev3dev should load the version I have.