ev3dev/ev3dev #1393: Does not connect to Open Roberta Lab

ppiech, 2020-03-16:
**What are you trying to do?**

Connect to Open Roberta Lab

**What did you expect to happen**

When I select `Open Roberta Lab` I expect to get the connection code.

**What actually happened**

I get the message `Not available`

**Source Code**

Here is the program I am trying to run.

```
PASTE PROGRAM HERE
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
macOS 10.14.6
Wifi (I can ssh to the device)

**Robot (please complete the following information):**
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534696E3
BOARD_INFO_TYPE=main
```

**Background Information**
I used a 2 year old version of ev3dev brickman 0.8.1, and connecting to open roberta.  I updated to this latest version and set up Wifi, but now the lab connection isn't working.  I'm able to ping lab.open-roberta.org from the brick.

ppiech, 2020-03-16:
I found the `/usr/bin/openroberta` executable and when I run it from ssh it lets me connect!  How do I configure it to run at startup?

ppiech, 2020-03-16:
Also when I try to run from the lab, the following program is downloaded but doesn't do anything.  Is open-roberta support RIP?
```
#!/usr/bin/python3

from roberta.ev3 import Hal
from ev3dev import ev3 as ev3dev
import math
import os

class BreakOutOfALoop(Exception): pass
class ContinueLoop(Exception): pass

_brickConfiguration = {
    'wheel-diameter': 5.6,
    'track-width': 18.0,
    'actors': {
    },
    'sensors': {
    },
}
hal = Hal(_brickConfiguration)

def run():
    hal.playTone(300, 100)
    hal.drawText("Hallo", 0, 0)
    hal.waitFor(5000)

def main():
    try:
        run()
    except Exception as e:
        hal.drawText('Fehler im EV3', 0, 0)
        hal.drawText(e.__class__.__name__, 0, 1)
        hal.drawText(str(e), 0, 2)
        hal.drawText('Press any key', 0, 4)
        while not hal.isKeyPressed('any'): hal.waitFor(500)
        raise

if __name__ == "__main__":
    main()
```

dlech, 2020-03-16:
> How do I configure it to run at startup?

```
sudo unmask openrobertalab.service # allows the service to run
sudo enable openrobertalab.service # enables the service at boot time
sudo start openrobertalab.service # starts the service right now
```

ppiech, 2020-03-16:
Got it!  It should've been 'sudo **systemctl** unmask'.

Open Roberta Lab works for me now :-)

closed 2020-03-16 by ppiech

dlech, 2020-03-16:
oops, thanks for the correction :smile: