ev3dev/ev3dev #1394: ALERT! /dev/mmcblk0p2 does not exist.

Krusoe, 2020-03-18:
Once I flashed an image for my ev3, it  does not load!
It outputs:"ALERT! /dev/mmcblk0p2 does not exist.Dropping to a shell.
I tried to reboot it,but it does not work
it's now in a sh shell with a prefix"(initramfs)"

WasabiFan, 2020-03-18:
Can you include what type of SD card you're using and what size it is? Make sure it meets the requirements here: https://github.com/ev3dev/ev3dev/wiki/Selecting-a-microSD-card. Also, if you have another SD card, try it as well.

Krusoe, 2020-03-18:
> Can you include what type of SD card you're using and what size it is? Make sure it meets the requirements here: https://github.com/ev3dev/ev3dev/wiki/Selecting-a-microSD-card. Also, if you have another SD card, try it as well.

Well, it seems like my card is 16Gigs and it's Class 10,made by SanDisk,is that OK?

dlech, 2020-03-18:
It should be OK, but even the best SD cards go bad.

Krusoe, 2020-03-18:
> It should be OK, but even the best SD cards go bad.

Thanks, I'll try another card later.

BernardoBotelho, 2020-05-21:
Hi. I got the same error message, but the difference is that I had been coding in python for over a week and everything was fine and suddenly it was not! I tried 3 different brand new SD cards, they all gave the same alert!

tedlvyi, 2020-05-21:
Hi. Same error here, two 16GB cards stop working and the same day.  EV3 light doesn't turn orange during the boot and stay off until the error shown on the display.  Is there anything to do with my EV3 block?

tedlvyi, 2020-05-21:
> It should be OK, but even the best SD cards go bad.
> 
> Thanks, I'll try another card later.

Hi Krusoe, have you resolved your issue?

tedlvyi, 2020-06-02:
My issue is resolved after fixing a lock pin of the SD card slot in EV3. If your EV3 light doesn't turn orange during the boot, consider your SD card slot, not the card itself.

ihormudryy, 2021-01-24:
@tedlvyi can you please give more details on how did you fix this issue in SD card slot because I have exactly the same problem?

dlotccjb, 2022-07-04:
> 

Follow your method, the problem is really solved! Thank you so much!

jluckenbach1, 2022-10-24:
Not sure if it will be as easy for others, but figured I would update this thread with what worked for me!
I took the EV3 apart and then inserted the SD card into the slot, and closed it back up.

Then it was finally able to boot up correctly!
So it was some sort of weird connection issue with the SD slot.

wkcochran123, 2025-05-15:
Anyone want to post a solution as well as indicating one exists?