ev3dev/ev3dev #1395: What USB hub would you recommend for EV3

ivan-marroquin, 2020-03-18:
**Question**

Hi all,

I am looking for a USB hub to connect a logitech 720 webcam and a wifi dongle into the EV3. Do you have some recommendations? I am using the latest version of ev3dev.

Many thanks,

Ivan

WasabiFan, 2020-03-23:
I don't personally have any suggestions here, but what I can say is that the USB port on the EV3 is only USB 1.1, so your webcam performance won't be very good. I expect any USB1-compatible hub (all of them, AFAIK) should work fine. A smaller one might be best for putting on mobile robots.

dwalton76, 2020-03-23:
I use these with a wifi dongle and webcam on an EV3:
https://www.amazon.com/Sabrent-4-Port-Individual-Switches-HB-UMLS/dp/B00BWF5U0M/ref=sr_1_5?keywords=usb+hub&qid=1584961710&sr=8-5

ivan-marroquin, 2020-03-24:
Hi @WasabiFan  and @dwalton76 

Many thanks for your suggestions!

Ivan

closed 2020-03-24 by ivan-marroquin