ev3dev/ev3dev #1397: Easiest way to control NXT slave from EV3 master?

LeeJenkins, 2020-03-29:
**Question**

Is it possible to control motors and sensors connected to a slave NXT from an ev3dev program running on a EV3 master?

I know that the standard LEGO EV3 software does not support this configuration. But I don't know if that because of HW incompatibility or because they just didn't implement it. From what I know (which isn't much), it appears that both bricks support the same bytecode communication protocol. But I must be missing something, because if it was truly identical, then it seems that the implementation would have been a no-op. So, there's that.

This question seems similar to "Does EV3Dev any LCP support #440" and "EV3 bytecode interperter/LEGO VM compatibility layer #329". But those issues seem to be about ev3dev executing LEGO bytecode, not using bytecode to control a remote NXT.

**Background Information**

I would like to use an NXT (with NXT sensors and motors) as a slave device attached to an EV3 master. Ideally, the NXT electronics would be instantiated in my ev3dev program as objects of the same classes as if they were connected directly to the EV3. If you take my meaning. The constructors would have a way to connect to the NXT slave port. The NXT (again, ideally) would be running its native FW.

This is another point where my understanding may be lacking. The LEGO NXT HW dev kit indicates that the NXT can be connected in a master/slave configuration over bluetooth. Ideally this same protocol coulda/woulda/shoulda been implemented over USB. But then again, maybe not.

I found a source code example using leJOS (which is not ev3dev, I realize).

    NXTMotor m = new NXTMotor(remoteNXT.getPort("A"));
    m.forward();
    Delay.msDelay(1000);
    m.stop();

source: https://lejos.sourceforge.io/forum/viewtopic.php?t=7137

I am trying to do this because I want to build a robot that uses lots of motors and sensors. I already own enough EV3 and NXT hardware to complete the project. But I don't know how to control the NXT slave's electronics from an ev3dev program.

**Screenshots**
Not applicable.

**Desktop (please complete the following information):**
Does not seem applicable. Will supply if needed.

**Robot (please complete the following information):**
 - Device: LEGO EV3 and NXT
 - Versions: [see below]

```
robot@ev3dev:~$ ev3dev-sysinfo -m
<!-- Copy everything between these lines -->
**System info (from `ev3dev-sysinfo`)**
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535492B5
BOARD_INFO_TYPE=main
<!-- Copy everything between these lines -->
```

dlech, 2020-03-29:
There is a Python library for remote controlling NXTs. https://github.com/Eelviny/nxt-python

To install on ev3dev:

    sudo apt update
    sudo apt install python3-nxt

LeeJenkins, 2020-03-30:
Thank you, David, nxt-python is awesome. I was able to install the package nxt-python-2.2.2 into a python 2.7 environment on linux. The example finds the NXT and turns the motors.

But then I tried installing python3-nxt per your instructions: first apt update, then apt install, except on my Ubuntu 18.04 laptop running python 3.7.6. But when I tried, apt printed this error: "E: Unable to locate package python3-nxt".

I'm a little confused. The nxt-python readme on github says the python3 is incomplete. There are no recent updates, so it appears abandoned. Further, a quick search for "python3-nxt" turns up empty. Would you expect the python3-nxt package to install on the ev3? Did I typo something?

All that said, I may be able to work with nxt-python-2.2.2 from the laptop. It's sub-optimal, but I need the laptop in the project for other reasons.

dlech, 2020-03-30:
My suggestion was to install it on the EV3 rather than a desktop. We made a special package for ev3dev.

LeeJenkins, 2020-03-30:
It worked, my bad. Thank you!

closed 2020-04-06 by LeeJenkins