ev3dev/ev3dev #1398: ev3 Button Class not working in R3 release

johnscary-ev3, 2020-03-30:
The Button Parameters Class does not seem to work in Release R3 compared to original LEGO release of micro-python.

Now when I use "Button.LEFT_UP"  to get the code for a button pressed in IR remote I get this error,

 File "/home/robot/EVE3RSTORM/main.py", line 412, in <module>
TypeError: unsupported types for __add__: 'Button', 'Button'

Line of Code with error:

 if butt_sum1 == Button.LEFT_UP or butt_sum1 == Button.LEFT_UP + Button.RIGHT_UP or butt_sum1 == Button.LEFT_UP + Button.RIGHT_DOWN:

johnscary-ev3, 2020-03-30:
After some investigation it looks like this statement to get the pressed buttons from the IR remote:

InfraredSensorPort = Port.S4
ir =InfraredSensor(InfraredSensorPort)
buttons = ir.buttons(chan)

now returns a list of literal button names instead a list of integer button codes as before.
Is this and intentional change or a bug?

dlech, 2020-03-30:
Hi, there is a new community support site for Pybricks MicroPython at https://github.com/pybricks/support. Would you mind raising the issue there?

johnscary-ev3, 2020-03-30:
Sure ok.
 Just put this issue in the other location. 
Thanks.

closed 2020-03-30 by johnscary-ev3