ev3dev/ev3dev #1400: PiStormsv2 - problem with `set_device`

KWSmit, 2020-04-05:
**Describe the bug**
Well, I think it's a bug...
In short, when you call `set_device` a second time on a port, the LEGO uart sensors will stop working. There is one exception, the GyroSensor doesn't have this problem.

**To Reproduce**
This can be reproduced in Python code, as well as in the console. That's why I think it's a kernel issue. I'll describe the Python way:

1. Boot PiStorms
2. Start Python REPL on the PiStorms (in Terminal via SSH)
3. Imports:

```
from  ev3dev2.sensor.lego import ColorSensor
from ev3dev2.port import LegoPort
```

4. Set port:

```
p = LegoPort('pistorms:BAS1')
p.mode = 'ev3-uart'
p.set_device = 'lego-ev3-color'
```

5. Right now you can see the sensor in Brickman's Device Browser. As expected after calling `set_device`.
6. Create sensor in code:

```
color_sensor = ColorSensor('pistorms:BAS1')
```

7. Now you can set the mode of the sensor and read values in code or in Brickman's Device browser. So far so good.
8. Set same port to 'none':

```
p.mode = 'none`
```

9. Sensor not visible any more in Device Browser, as expected.
10. Set port:

```
p.mode = 'ev3-uart'
```

11. Now, the latest sensor used on this port is visible again in Brickman's Device Browser, in this example the ColorSensor and you can see the sensor is active (sensor's light turns on). You can even read values in Device Browser. That is not what I would expect. I would expect it only to be visible after `set_device` is called.
12. Call `set-device`:

```
p.set_device = 'lego-ev3-color`
```

13. This makes the sensor turn off (lights go out) and there are no readings. In Brickman's Device Browser you see a fixed value, most of the time the last value it read. The same happens in Python code.

**Why is this a problem?**
There is no autodetection on PiSrorms, so you have tot set port and sensor manually in the code, the way as described above. When you run your script the first time after boot of the system, it works fine, but not when you start the script a second time. Or when you first run a script which uses the ColorSensor and after that start a script with a different sensor on the same port.
The port seems to have some kind of memory which sensor was used last (until the system is rebooted) and sets the device last used automatically when port mode is set to `ev3-uart`. When you call `set_device` again when this is the case the sensor stops working.
I'm having this problem with all LEGO uart-sensors, except the GyroSensor. That works fine all the time. In above description the sensor is not visible in step 11 (when port mode is set to `ev3-uart`), as I would expect. It's only visible after `set_device` is called. So, it seems the port does not 'remember' the GyroSensor.

Sorry for the long explanation,I hope it's clear what I mean.

**Expected behavior**
Mentioned above.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Connection Type: Wi-Fi
 - Other software: Visual Studio Code or Terminal on PiStorms via SSH.

**Robot (please complete the following information):**
 - Device: PiStormsv2
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-rpi2-generic-2020-03-07
Kernel version:     4.14.114-ev3dev-2.3.5-rpi2
Brickman:           0.10.3
BogoMIPS:           38.40
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000bc6aa21f
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=V2.08
BOARD_INFO_MODEL=PiStorms
BOARD_INFO_TYPE=aux

```

dlech, 2020-04-05:
As a workaround, could you set the mode of the port to `none` and then set it back to `ev3-uart`?

KWSmit, 2020-04-05:
Sure, but more important don't call `set_device`. It's not even necessary to set the port to `none` and then back again to `ev3-uart`. So, in the Python code I have to check if the sensor is already there. There's no Python function to do that, but I guess I could check the filesystem (check if directory `pistorms:BAS1:lego-ev3-color` exists for the port it's attached to). But if I like to use a different sensor on the same port, I have to reboot the system first.
Btw, do you have any idea why the GyroSensor behaves differently? That means in this case, behaves properly.
At the moment it's no problem for me, I'm not working on a project right now. I was just testing.

KWSmit, 2020-04-05:
Just got an idea: if I attach another sensor I have to check if the corresponding directory for that sensor exists (for instance `pistorms:BAS1:lego-ev3-us`) That should work. So an all-round solution for this moment is:
1. set port mode
2. check for directory `pistorms:XXXX:lego-ev3-yy (XXXX is port to which sensor is connected, eg. BAS1 and yy is sensor).
3. If the directory is not there, use `set_device` or else don't.
A quick test showed that this works.

KWSmit, 2020-04-06:
This works for me:

```
#!/usr/bin/env python3
import sys
import os
from time import sleep
from ev3dev2.sensor.lego import ColorSensor
from ev3dev2.port import LegoPort
from ev3dev2.button import Button

button = Button()

address = 'pistorms:BAS1'
p = LegoPort(address)
p.mode = 'ev3-uart'
sleep(0.5)
path_name = '/sys/devices/platform/soc/3f804000.i2c/i2c-1/1-001a/lego-port/port4/pistorms:BAS1:lego-ev3-color'
if not os.path.exists(path_name):
    p.set_device = 'lego-ev3-color'

s = ColorSensor(address)
s.mode = ColorSensor.MODE_COL_COLOR
while not button.enter:
    print(s.color_name, file=sys.stderr)
    sleep(0.1)
```

I wrote the same script also for `UltrasonicSensor` and can use the scripts alternately with succes.