ev3dev/ev3dev #1403: Actounix L12 100 not recognized

trinidadg, 2020-04-14:
I've been working with ev3-dev for the past weeks and successfully configured various sensors and actuators. However I can't get the Actuonix Linear Actuator working.

When I try to initialize the class, it says the motor is not connected.

I checked for the driver and it is there:

```
robot@ev3dev:~$ cat /sys/bus/lego/drivers/ev3-motor/driver_names 

lego-nxt-motor lego-ev3-l-motor lego-ev3-m-motor act-l12-ev3-50 act-l12-ev3-100
```

I have changed the port and the cable just in case and had no luck.

When I create a simple Motor class in that port It uses the lego-ev3-l-motor driver.
Using that it responds to the on(vel) command, going both ways and stopping on the edges. I can't use stop() (it does nothing) and when asked for position it says 0 or 1 being fully retracted or fully extended.

Really hope for some help! Thanks 

**Desktop :**
 - OS and version: Ubuntu 18.04
 - Connection Type: Wi-Fi
 - Software: ssh

**Robot:**
 - Device: Ev3

ev3dev-sysinfo -m output:
```
Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165365FFCA
BOARD_INFO_TYPE=main
```

dlech, 2020-04-14:
Using a `LegoPort` object it is possible to override the detected device. There is a similar example for how to do this with sensors at https://github.com/ev3dev/ev3dev-lang-python-demo/blob/stretch/platform/brickpi3-motor-and-sensor.py.

In your case `port.set_device = 'act-l12-ev3-100'` should do the trick.

trinidadg, 2020-04-14:
Hi, thank you for your immediate answer! 

I created this scipt:
```
#!/usr/bin/env python3
from ev3dev2.motor import ActuonixL12100Motor, OUTPUT_C
from ev3dev2.port import LegoPort

p = LegoPort(OUTPUT_C)

p.set_device = 'act-l12-ev3-100'

motor = ActuonixL12100Motor(OUTPUT_C)

```

But when i run it i get this error:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 244, in _get_attribute
    attribute = self._attribute_file_open( name )
  File "/usr/lib/python3/dist-packages/ev3dev2/__init__.py", line 227, in _attribute_file_open
    mode = stat.S_IMODE(os.stat(path)[stat.ST_MODE])
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/tacho-motor/linear9/count_per_rot'

```

And results in:
```
ev3dev2.DeviceNotFound: ActuonixL12100Motor(ev3-ports:outC) is no longer connected
```

dlech, 2020-04-14:
Try adding a time delay between `set_device` and `ActuonixL12100Motor` as in the linked example.

trinidadg, 2020-04-14:
Tried that and I keep getting the same error. Changed the time delay up to 20 seconds just in case it was being slow, but nothing.

dlech, 2020-04-14:
> FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/tacho-motor/linear9/count_per_rot'

OK, I think this is a bug in the python library then. `linear9` should probably be `motor9`. You can check for sure in a SSH terminal.

    $ ls /sys/class/tacho-motor/

WasabiFan, 2020-04-14:
Is the `count_per_rot` attribute omitted for linear actuators? If not, it definetely looks like a library bug somehow.

dlech, 2020-04-14:
It's been a long time. My memory is probably faulty :wink: Better check the docs: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/motors.html#tacho-motor-subsystem

So maybe `linear` is right and the attributes are slightly different.