ev3dev/ev3dev #1404: Ev3 is not listening on port 22

maras75, 2020-04-22:
**What are you trying to do?**

I am trying to connect from MS Visual Studio Code. finally I used nmap to scan Ev3 ports,

**What did you expect to happen**
Connection through MS Code and SSH

**What actually happened**
Nmap doessn't show port 22 opened or even closed

**Desktop (please complete the following information):**
 - OS and version: Win 8.1
 - Connection Type: USB
 - Software: Visual Studio Code, PuTTY

**Robot (please complete the following information):**
 - Device: EV3,
 - Versions: ev3dev-stretch-ev3-generic-2020-04-10,  ev3dev-stretch-ev3-generic-2019-10-23

**Background Information**

I'm able to ping Ev3. USB and BT work correctly on built-in Ev3 system.

dlech, 2020-04-23:
We've had a few people in the past report that the SSH server didn't start because of a defective microSD card.

maras75, 2020-04-25:
Ok problem was solved. I have specific application installed (to control plugged external devices). After uninstalling connection via USB and BT works perfectly :)

closed 2020-06-23 by dlech