ev3dev/ev3dev #1406: Is it possible to read image data programmatically from the Pixy camera?

LeeJenkins, 2020-05-03:
**Question**

Is it possible to read image data programmatically from the Pixy camera?

There is a windows program PixyMon (also in qt/linux) that will display live images from the camera over USB, but I cannot find any documentation for reading an image programmatically. I read the EV3 pixy tutorials at https://www.ev3dev.org/docs/tutorials/using-pixy-camera/, but the examples are all about getting the (x,y,height,width) of a centroid through the I2C port interface. I want to read an image and I assume this will be over USB.

It would be ideal if the interface was in python. But I could work with a C/C++ API, command line app, or reading raw data from a device in /dev.

**Background Information**

I am trying to read an image because I want to build a robot that reads a short barcode label on an object on a conveyor. The label has a colored dot for pixy centroid tracking. The colored dot allows the robot to stop the conveyor when an item (and its barcode) are in position. But the program needs to read a barcode to identify the object on the conveyor.

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
N/A

**Robot (please complete the following information):**
 - Device: LEGO EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Image file:         ev3dev-stretch-ev3-generic-2019-03-03
Kernel version:     4.14.96-ev3dev-2.3.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535492B5
BOARD_INFO_TYPE=main
```

dlech, 2020-05-03:
ev3dev ships with libpixyusb which include python bindings. This looks like a good place to start: http://www.cmucam.org/boards/9/topics/4451

LeeJenkins, 2020-05-03:
That seems to be promising, but I have hit a bit of a dead end. The forum post you linked discusses a C function `pixy_command`, using the string command name `"cam_getFrame"`. There was a further link to http://www.cmucam.org/attachments/1094/hello_pixy.cpp, which is an example c++ program that uses `pixy_command("cam_getFrame",...)` to retrieve an image. That seemed promising, except `hello_pixy.cpp` includes `pixy.h`, which is not installed on ev3dev. (I ran ` find /usr -name pixy.h`, which came up empty.)

Next I tried using the python pixy package. The program imported `pixy` and dumped introspection info about the `pixy` module and its sub-module `_pixy`. I found python bindings for retrieving (x,y,height,width), but nothing for images. 

Out of curiosity I checked to see if the `pixy_command` function existed anywhere in the python package. The symbol appears in the .so, but not in any of the python source:

```
   robot@ev3dev:~/research-lab$ grep pixy_command -R /usr/lib/python3/dist-packages/
   Binary file /usr/lib/python3/dist-packages/_pixy.cpython-35m-arm-linux-gnueabi.so matches
```

So I don't know if it's exposed through the python API or how to access it. If I had the C source for the python package, it might have enough information for me to learn how to make the call.

In other searching, I also found a project called discoverypixy (https://github.com/t-moe/discoverpixy), a C++ app that is supposed to retrieve pixy image data on linux. But the code does not build properly on ev3dev. It has its own `hello_pixy.cpp` source. I spent an hour or so working around compiler errors. I built libpixy.a, only to find it has dependencies in the project's USB library, which also doesn't compile. I can see that compiling this code is going to be a huge time sink, but right now it's my only semi-viable path.

Finally, I also tried building the source from charmed labs (http://charmedlabs.github.io/pixy/index.html). But gcc spends hours compiling code, only to exit with "error 4". I'm guessing some kind of memory issue, but I'm not going to root cause it.

It seems like I'm soooo close. Do you know of anyone who has successfully read image data from the pixy on ev3dev? With example source?

dlech, 2020-05-04:
The python package uses SWIG to provide bindings to the C++ library. In other words, it is just a thin wrapper around the C++ library rather than a "real" python library. So the .i file is basically the source for the python library. Unfortunately it looks like pixy_command is not included there.

The ev3dev package source is here: https://github.com/ev3dev/pixy/.

For compiling C++ code for ev3dev we recommend using docker to cross-compile https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/. In the docker environment (and on the EV3) you should be able to `sudo apt update && sudo apt install libpixyusb-dev:armel` to get the `pixy.h` header file.

LeeJenkins, 2020-05-07:
@dlech 

I decided that modding the PixyCam package was a little bigger bite than I was willing to chew. I ended up using an old USB webcam and writing some filters to duplicate the functionality of the PixyCam. If you're interested, you can read about it here: https://eltiriel.wordpress.com/2020/05/07/the-better-to-see-you-with/

I appreciate your assistance. I may yet find a compelling reason to set up the docker cross-compiler image. But not today. :-)

closed 2020-05-07 by LeeJenkins

dlech, 2020-05-08:
nice blog post :smile: