ev3dev/ev3dev #1407: 64GB SanDisk Ultra works

NikosBricks, 2020-05-03:
**Question**

Is it still 32GB SD cards the max.?

**Background Information**
I had a SanDisk Ultra 64GB SD card sitting around and I thought trying it and on my surprise it worked!
Through an ssh session I have captured the df output of my EV3 with the Sandisk Ultra 64GB SD:
robot@ev3dev-A:~$ uname -a;df -h /dev/mmcblk0p2
Linux ev3dev-A 4.14.117-ev3dev-2.3.5-ev3 #1 PREEMPT Sat Mar 7 12:54:39 CST 2020 armv5tejl GNU/Linux
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   59G 1017M   56G   2% /

Interestingly enough it only reports 59GB of the 64GB.

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: 4.14.117-ev3dev-2.3.5-ev3

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          4.0
Board:              board0
BOARD_INFO_HW_REV=8
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=7
BOARD_INFO_SERIAL_NUM=00165380DBA2
BOARD_INFO_TYPE=main
```

R2D2Hud, 2020-05-12:
I had the same thing with a 64GB card

NikosBricks, 2020-05-12:
Thanks R2D2Hud for confirming this. Mine it still working fine.

Does anyone knows why the 32GB USB limit?

dlech, 2020-05-12:
According to the specification, the SoC in the EV3 is SDHC. 

https://www.ti.com/product/AM1808

---

SanDisk has mixed messages about compatibility:

> SDXC cards will work in SDHC compatible readers (not SD readers) if the computer OS supports exFAT. 

> SDXC memory cards will NOT work with SD / SDHC host devices.

https://kb.sandisk.com/app/answers/detail/a_id/2520/~/sd/sdhc/sdxc-specifications-and-compatibility

---

Maybe they are doing a good job making them backwards compatible even if it is not officially supported? Although I wonder if there is an address limit, e.g. what happens if a file is placed at >32GB offset on the card?

NikosBricks, 2020-05-12:
All good points. As they say, "your mileage may very"! Also, very few users will need more than 32GB on-board an ev3.

When I have a chance I will try to create a file larger than 32GB and see what happens. Let you know.

dlech, 2024-05-10:
Probably best to keep it safe and go with 32GB. I highly recommend https://www.westerndigital.com/products/memory-cards/sandisk-extreme-uhs-i-microsd?sku=SDSQXAF-032G-GN6MA