ev3dev/ev3dev #1412: Cannot Bluetooth-connect with EV3 device in VSCode EV3Dev Device Browser on 1 particular MacBook

TheVinhLuong102, 2020-06-07:
**What are you trying to do?**
I am trying to Bluetooth-connect with the EV3 brick via the Visual Studio Code EV3Dev Device Browser, and am unable to do so with my MacBook (MacBook Pro, Mid 2015) running MacOS Catalina.

**What did you expect to happen**
I have actually succeeded to make the connection with several *other* very similar MacBooks (e.g. my wife's), which all had similar VSCode and the EV3Dev Device Browser plug-in installed.

**What actually happened**
However, on my own MacBook, I keep failing to make the Bluetooth connection via the VSCode Device Browser. The error messages are captured in the screenshots attached.

"Failed to connect to ev3dev: connect EHOSTUNREACH fe80::217:e9ff:feb1:9303%*undefined*:22 - Local (:::<some-number>)"

[image]

I notice the word "*undefined*" in the error message and it seems to be related to the problem.

dlech, 2020-06-07:
Does this help? https://github.com/ev3dev/vscode-ev3dev-browser/wiki/Troubleshooting

TheVinhLuong102, 2020-06-07:
@dlech thank you. I followed the troubleshooting guide above but that has not resolved the problem. The error messages like those in the screenshots persist -- even if the device appears as "Connected" in the System Preferences >> Bluetooth screen:

[image]

The screenshots of my System Preferences >> Network settings are as follows:
- for Wifi: 
[image]

- for Bluetooth PAN:
[image]

Is there something that I am missing? Thank you very much, @dlech.

TheVinhLuong102, 2020-06-07:
By the way, @dlech one thing that is different between my case and the troubleshooting scenario described in https://github.com/ev3dev/vscode-ev3dev-browser/wiki/Troubleshooting is that I see the "undefined:22". I'm not sure what that means. Does that look symptomatic of some known problem?

dlech, 2020-06-07:
`undefined` is supposed to be the name of the network interface. Maybe try removing the "Bluetooth PAN" device and set it up again. Also try turning off Internet Sharing and turning it on again. And try removing the Bluetooth device from Bluetooth as well and try pairing it again.

TheVinhLuong102, 2020-06-07:
@dlech ah, I have just restarted my laptop to make sure the setting changes you specified get applied, and the connection is working now! :) Maybe you can consider adding Rebooting the Mac to the advice given in https://github.com/ev3dev/vscode-ev3dev-browser/wiki/Troubleshooting#macos?

This issue is now resolved. Thank you so much, @dlech !

dlech, 2020-06-07:
Great. Rebooting fixes so many problems. :smile:

TheVinhLuong102, 2020-06-07:
Many thanks again @dlech :)

closed 2020-06-07 by TheVinhLuong102