ev3dev/ev3dev #1413: Custom sensor kernel driver

glingy, 2020-06-07:
**Question**

I just recently created a custom sensor to use with the ev3 (i2c) and it follows the lego sensor conventions and is recognized correctly by the ev3 as an nxt-i2c sensor. I wanted to add a kernel driver for it so I can use it as a supported sensor, and I have never done that before. 

I found the link on the contribute page https://www.ev3dev.org/contribute/ to an example pull request, but I cannot find how to just build it myself. Would I need to rebuild the entire kernel to add an i2c sensor or is it modular and I just have to replace something somewhere with a rebuilt version? I could probably figure out how to make the sensor work given the example pull request, but I can't find a guide anywhere on how to build it. 

Specifically, if it matters, it's a board I made with a TI MSP430 that will respond to commands (writes to registers 0x50 through 0x53, each one byte) and allow the ev3 to read the registers 0x43 and the same 0x50 through 0x53. I got the processor to work and be recognized as a hitechnic infrared receiver by sending the same product and vendor identification information, so I know that much is working correctly.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS and version: macOS 10.15
 - Connection Type: N/A
 - Software: N/A

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch (I think) kernel: 4.14.117-ev3dev-2.3.4-ev3

Thanks for the help!

dlech, 2020-06-08:
If this is a one-of-a-kind sensor, then it probably doesn't make much sense to create a kernel driver for it. It would be easier to just use direct I2C commands to communicate with it.

https://github.com/ev3dev/ev3dev-buildscripts has instructions on how to build the kernel if you want to try it anyway.

glingy, 2020-06-08:
I just found that link separately and was about to add a duh reply because it's practically exactly what I was asking about. And yes, I could use direct I2C, but I thought it might be a good experience to try doing it the hard way and see if I could get it working. Thanks for the help, we'll see how this goes.

closed 2020-06-13 by glingy