ev3dev/ev3dev #1418: Installing Python 3.8.3 on EV3

FlorianRaediker, 2020-06-19:
**What are you trying to do?**

I am trying to install the newest Python version, 3.8.3. 
I followed the guide here: https://docs.python-guide.org/starting/install3/linux/ as I did not want to build Python from source on the EV3...

**What did you expect to happen**

`sudo add-apt-repository ppa:deadsnakes/ppa` should work properly.

**What actually happened**

`add-apt-repository` does not work:

```
$ sudo add-apt-repository ppa:deadsnakes/ppa
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 95, in <module>
    sp = SoftwareProperties(options=options)
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 109, in __init__
    self.reload_sourceslist()
  File "/usr/lib/python3/dist-packages/softwareproperties/SoftwareProperties.py", line 599, in reload_sourceslist
    self.distro.get_sources(self.sourceslist)    
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 93, in get_sources
    (self.id, self.codename))
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for ev3dev/stretch
```
(PS: The same error occurres when running `sudo apt-add-repository ppa:ev3dev/tools` from https://www.ev3dev.org/docs/devtools/installing-the-ev3dev-archive/ )

**Desktop:**
 - OS and version: Windows, macOS
 - Connection Type: Wi-Fi
 - Software: ssh

**Robot:**
 - Device: EV3
 - Versions: 

```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=XXXXXXXXXXXX
BOARD_INFO_TYPE=main
```

**Background Information**
I'm not sure whether installing Python this way is the best idea - are there any other options to get the newest Python version?

Many thanks in advance.

martia86, 2020-06-24:
don't try to install python3,i think it has python3 and python by default.
also if you want to install python yourself,first do:
`sudo apt-get update`
and
`sudo apt-get upgrade`

hope it helps! 😉

FlorianRaediker, 2020-06-24:
Thanks for your comment, I know Python 3 is already installed, but it's 3.5, not the newest version 3.8.3 (Debian stretch repositories only have Python 3.5 from what I know). 
I did of course try `sudo apt-get update/upgrade`, but as no repositories have Python 3.8, something like `sudo apt-get install python3.8` won't help (the package doesn't exist). 
That's why I tried to add a repository which has Python 3.8 (`sudo add-apt-repository ppa:deadsnakes/ppa`), but as I said, adding the repository doesn't work.

JorgePe, 2020-07-03:
Packages from deadsnakes/ppa are built for x64 architecture (Intel and AMD processors) not for armel (EV3) or even armhf (Raspberry Pi).

FlorianRaediker, 2020-07-04:
> Packages from deadsnakes/ppa are built for x64 architecture (Intel and AMD processors) not for armel (EV3) or even armhf (Raspberry Pi).

Oh, I didn't know that, thanks. So the only way would be to build Python 3.8 from source? Or are there any packages for armel?

JorgePe, 2020-07-06:
I didn't look much for it... python is an important part of Debian, also of ev3dev, so lots of packages depend of it... even if you find all required packages already built for armel (something that I really doubt outside of Debian, armel being so old) you might risk damaging the distro as a whole.
Debian does have python3.8 for armel (https://packages.debian.org/bullseye/python3.8) but i think bullseye is something like the next Debian release after buster... and main ev3dev is still based on stretch.

May I ask why do you require 3.8 specifically?

FlorianRaediker, 2020-07-06:
> May I ask why do you require 3.8 specifically?

It just would be good if one could use the newest Python features: f-strings (since 3.6) walrus operator (since 3.8), proper asyncio, etc.
(So it's not like I need to use a specific module which is only compatible with a newer version, the only reason I would like to use Python 3.8 is newer syntax.)

Thanks for your help, I'll just use Python 3.5 then.

Nikolay-Zagrebin, 2024-06-04:
> **What are you trying to do?**
>
> I am trying to install the newest Python version, 3.8.3. I followed the guide here: https://docs.python-guide.org/starting/install3/linux/ as I did not want to build [...]

At the moment, experimenting for a long time, I managed to install python 3.10.14. I had to suffer for a long time with updating openssl to version 1.1.1.
Next, I plan to experiment with installing 3.11 and 3.12.

Nikolay-Zagrebin, 2024-06-04:
> It just would be good if one could use the newest Python features: f-strings (since 3.6) walrus operator (since 3.8), proper asyncio, etc. (So it's not like I need to use a specific module which is [...]

I managed to upgrade to python version 3.10.14, I'm just finishing the tests.

Madelyn-of-Hell, 2024-06-19:
> At the moment, experimenting for a long time, I managed to install python 3.10.14. I had to suffer for a long time with updating openssl to version 1.1.1. Next, I plan to experiment with installing 3.11 and 3.12.

i'm trying to get 3.12 on mine currently, would you mind explaining how you did it?

Nikolay-Zagrebin, 2024-06-19:
> i'm trying to get 3.12 on mine currently, would you mind explaining how you did it?
we will discuss it in more detail here
https://github.com/ev3dev/ev3dev/issues/1635#issuecomment-2177876214