ev3dev/ev3dev #1420: Problem booting ev3dev on an EV3 brick

FarukIb, 2020-06-23:
**What are you trying to do?**

I am trying to boot ev3dev on my EV3 brick.

**What did you expect to happen**

I expected it to boot up without any issue.

**What actually happened**

At first I couldn't boot ev3dev but I got a new SD card so I managed to make it work. I wasn't done with setting up the EV3 and my computer to start coding when something went wrong. I decided that the best course of action would be to reflash the SD card but when I did it stopped working. I got the message:
 **_"Gave up waiting for root file system device. Common problems ..."_** and
_**"ALERT! /dev/mmcb1k0p2 does not exist. Dropping to a shell!".**_
 I tried to boot my Raspberry Pi 3 from the SD card and it worked. I also tried booting Lejos EV3 on the brick and that also worked. I tried updating the firmware on the brick and it didn't help. I also tried flashing the SD card using different software and flashing it on a linux system. And the last thing I tried was using older versions of ev3dev.

**Desktop:**
 - OS and version: Windows 10 1903 & Linux Mint 19.3 Cinnamon 64-bit booted from a 16GB USB stick

**Robot (please complete the following information):**
 - Device: EV3

dlech, 2020-06-23:
You have been very thorough in your troubleshooting. :+1: 

Perhaps it could be a hardware problem like https://github.com/ev3dev/ev3dev/issues/894#issuecomment-305368697?

FarukIb, 2020-06-23:
Forgive me if I'm wrong but I don't see it being a hardware problem because Lejos EV3 booted just fine. And if it was a hardware problem what could I do seeing as I don't have spare bricks.

martia86, 2020-06-24:
Hello,
Try Reformating Your SD,or make a smaller partition.

i hope it helps!
😉

FarukIb, 2020-06-29:
Sorry for responding late. I tried both of those and again nothing changed.

prakharg01, 2020-07-18:
> **What are you trying to do?**
>
> I am trying to boot ev3dev on my EV3 brick.
>
> **What did you expect to happen**
>
> I expected it to boot up without any issue.
>
> **What actually happened**
>
> At first I [...]

I am experiencing a similar problem and my ev3 leds are not turning orange during booting. it cant be a hardware problem as my brick is recognizing external sd cards in the normal ev3 lego firmware 

can someone help?

ihormudryy, 2021-01-23:
I've had exactly the same problem and the weirdest thing that this behavior is non-deterministic. Initially, it was working perfectly fine and suddenly it started failing on boot. I have solved it by booting from LejOS 7.7 image with the same SD card.  https://sourceforge.net/projects/ev3.lejos.p/files/

dlotccjb, 2022-07-04:
> 
https://github.com/ev3dev/ev3dev/issues/1394#issuecomment-637263812