ev3dev/ev3dev #1424: Unable to copy files from desktop to EV3 using scp command in ubuntu

ivan-marroquin, 2020-07-09:
**What are you trying to do?**
I would like to copy *.wav files from my desktop to the EV3 by using scp command. 

From the Visual Studio Code window, I noticed that EV3DEV DEVICE BROWSER section, the device is listed as ev3dev-4. Then, if I open a ssh connection from Visual Studio Code the following prompt is shown robot@ev3dev

I am trying to ...?

On my desktop, I open a linux terminal and change directory to folder with *.wav files to be copied to EV3. Then, I tried these commands:

a) scp fear_not.wav robot@ev3dev.local:/home/robot/sounds

ssh: Could not resolve hostname ev3dev.local: Name or service not known
lost connection

b) scp fear_not.wav robot@ev3dev.local:/home/robot/sounds

ssh: Could not resolve hostname ev3dev-4.local: Name or service not known
lost connection

I also tried to ssh to ev3 using the command in https://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/
and I still have connection issue:

ssh robot@ev3dev.local
ssh: Could not resolve hostname ev3dev.local: Name or service not known

Note that in Visual Studio Code window, python scripts can be sent to EV3 without issues.

**What did you expect to happen**

I would expect the connect to properly work, so then the *.wav files can copied from my desktop onto EV3. Any suggestions?

**What actually happened**

The program did...

**Source Code**

Here is the program I am trying to run.

```
PASTE PROGRAM HERE
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: Ubuntu 18.04.4 LTS
 - Connection Type: SSH
 - Software: Ubuntu

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```
Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653555C5E
BOARD_INFO_TYPE=main

**Background Information**

I just want to copy *.wav files from my desktop to EV3 to play these sounds in my Python scripts.

dlech, 2020-07-10:
If you copy the sound files to your Python project folder, then they will get copied at the same time the python files get copied.

If you don't set up internet connection sharing or a manual IPv4 address on the EV3, by default ev3dev will only use IPv6 with a local-link address. For some reason, Linux doesn't know how to properly resolve local link IPv6 addresses which causes `ssh` and `scp` to fail.

- https://blogs.gentoo.org/eva/2010/12/17/things-you-didnt-known-about-ipv6-link-local-address/
- https://superuser.com/questions/1086954/how-do-i-use-mdns-for-ssh-6
- https://bugs.launchpad.net/ubuntu/+source/nss-mdns/+bug/369008

ivan-marroquin, 2020-07-11:
Hi @dlech 

Many thanks for your suggestions. I noticed that the IP address in EV3 changed. So, I tried this command: scp <file name> robot@<IP address>:~/sounds 

and it worked as expected. Could you explain how to configure a manual IPv4 address on the EV3?

Kind regards,

Ivan