ev3dev/ev3dev #1425: Using NXTCam-v4 with ev3dev and python

Pascal567, 2020-07-12:
I'm trying to use the [NXTCam-v4](http://www.mindsensors.com/ev3-and-nxt/14-vision-subsystem-camera-for-nxt-or-ev3-nxtcam-v4) for my EV3. I'm using ev3dev and python. The camera is a [supported sensor](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#mindsensors-com-vision-subsystem-v4-for-nxt-or-ev3) for ev3dev.

I calibrated the colors with the NXTCamView Software and uploaded them to the NXTCam. The "Color tracking mode" in the Software works fine but I'm not able to get the camera working with my EV3:
```
from ev3dev2.sensor import Sensor

cam = Sensor("ev3-ports:in2:i2c1")  # Camera on Port2
while True:
    debug_print(cam.value(0))
```
This prints zeros in the console. No matter what the camera sees. In my understanding, however, it should show the number of objects with the specified color.

The EV3 auto-detects the camera and it seems to work properly.
I'm very new to ev3dev. What am I doing wrong?

gara-3004, 2023-04-02:
Hi I am having the same problem. Does somebody have found a solution?

Almondus88, 2023-05-10:
Hello! I had the same problem. It seems like the NXTCam needs a command to begin tracking. The following code worked for me:

#!/usr/bin/env python3

import os
import sys

from ev3dev2.sensor import INPUT_1
from ev3dev2.sensor import Sensor

cam = Sensor("ev3-ports:in1:i2c1")  # Camera on Port1

cam.command = 'TRACK-ON'  #the command used to begin tracking