ev3dev/ev3dev #1426: Brickman go to 60% cpu each time you navigate with a button

freezed-or-frozen, 2020-07-13:
**Question**
Brickman consumes a lot of CPU each time you press a button to navigate into menus. How can I prevent brickman to be so cpu consuming ? Navigation with brickman was much more fluid in older image.

Thank you in advance for all your answers.

**Background Information**
I used to work with EV3 with students for a few years now, mainly for Python and C++ development. Because it was an old image installed on the robot (ev3-ev3dev-jessie-2015-12-30.img.zip), I try to install a new one : ev3dev-stretch-ev3-generic-2020-04-10.zip

But the first impression after boot is that the robot is very slow. Each time you navigate with the button into the menu, brickman go to 50-70% CPU. Maybe this issue is well known. Issue #1022 seems to talk about it. It is now closed and I do not find any answer.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS and version: Ubuntu 18.04
 - Connection Type: Wifi
 - Software: ssh

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: ev3dev-stretch-ev3-generic-2020-04-10.zip

```
<!-- Copy everything between these lines -->
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165346EE00
BOARD_INFO_TYPE=main
```
<!-- Copy everything between these lines -->

```

dlech, 2020-07-13:
I think the main issue could be seeing is the fact that text rendering has poor performance. https://github.com/ev3dev/grx/issues/17

Also, in general, newer software just runs slower on older devices.

freezed-or-frozen, 2020-07-14:
I have tested the official Lego image, which use the same kernel version and brickman works like a charm. I don't know the difference between the 2 images, but the Lego one works pretty well.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3-micropython-v2.0.0-sd-card-image
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           148.88
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535C4206
BOARD_INFO_TYPE=main
```

dlech, 2020-07-14:
Basically, the difference it branding. So I wouldn't expect any difference in behavior in this regard. Was anything else different (SD card, EV3 brick, installed files, attached motors/sensor, connection method, etc)?

freezed-or-frozen, 2020-07-16:
Something strange appears. As the first boot of the lego image seems to work pretty well (especially brickman), the second boot reveals the same issue as before : very slow navigation with brickman up to 70% cpu each time you push a button