ev3dev/ev3dev #1429: EV3DEV device browser not working

uelundgren, 2020-07-29:
**What are you trying to do?**

I am trying to connect to an EV3 device for the first time

**What did you expect to happen**

When clicking on "click here to connect to a device" in Visual Studio Code, I expected a window to pop-up, letting me choose my device

**What actually happened**
Nothing at all. 

The program did...

**Source Code**

Here is the program I am trying to run.

```
PASTE PROGRAM HERE
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS and version: MacOS 10.15.6
 - Connection Type: USB / Bluetooth
 - Software: Visual Studio Code 

Version: 1.47.3
Commit: 91899dcef7b8110878ea59626991a18c8a6a1b3e
Date: 2020-07-23T13:08:29.692Z (6 days ago)
Electron: 7.3.2
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Darwin x64 19.6.0

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: v.2.0.0

Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
 Where do I find this?

dlech, 2020-07-29:
It is working for me.

macos 10.15.6
vscode: 1.47.3
ev3dev-browser: 1.2.0

[image]

uelundgren, 2020-07-30:
Thanks for the info! Ok, so it would be an error specific to my computer, then, I guess? I tried logging in as a different user - no change, and tried reinstalling -no change.

Here is the log entry created when I press “Click here to connect to a device”:

log.ts:197   ERR No application in the Launch Services database matches the input criteria.: Error: No application in the Launch Services database matches the input criteria.
log.ts:197   ERR connect ENOENT /var/run/dbus/system_bus_socket: Error: connect ENOENT /var/run/dbus/system_bus_socket
	at PipeConnectWrap.afterConnect [as oncomplete] (net.js:1056:14)
console.ts:137 [Extension Host] rejected promise not handled within 1 second: Error: Request error
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:402
_invokeHandler @ rpcProtocol.ts:387
_receiveRequest @ rpcProtocol.ts:303
_receiveOneMessage @ rpcProtocol.ts:230
(anonymous) @ rpcProtocol.ts:105
fire @ event.ts:587
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
(anonymous) @ ipc.net.ts:592
fire @ event.ts:587
acceptChunk @ ipc.net.ts:239
(anonymous) @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:203
addChunk @ _stream_readable.js:295
readableAddChunk @ _stream_readable.js:276
Readable.push @ _stream_readable.js:210
onStreamRead @ internal/stream_base_commons.js:166
console.ts:137 [Extension Host] stack trace: Error: Request error	at Service.<anonymous> (/Users/erik/.vscode/extensions/ev3dev.ev3dev-browser-1.2.0/out/dnssd/dnssd-client.js:773:27)	at Generator.next (<anonymous>)	at fulfilled (/Users/erik/.vscode/extensions/ev3dev.ev3dev-browser-1.2.0/out/dnssd/dnssd-client.js:36:58)	at processTicksAndRejections (internal/process/task_queues.js:85:5)
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:402
_invokeHandler @ rpcProtocol.ts:387
_receiveRequest @ rpcProtocol.ts:303
_receiveOneMessage @ rpcProtocol.ts:230
(anonymous) @ rpcProtocol.ts:105
fire @ event.ts:587
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
(anonymous) @ ipc.net.ts:592
fire @ event.ts:587
acceptChunk @ ipc.net.ts:239
(anonymous) @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:203
addChunk @ _stream_readable.js:295
readableAddChunk @ _stream_readable.js:276
Readable.push @ _stream_readable.js:210
onStreamRead @ internal/stream_base_commons.js:166
mainThreadExtensionService.ts:65 [[object Object]]Request error
$onExtensionRuntimeError @ mainThreadExtensionService.ts:65
_doInvokeHandler @ rpcProtocol.ts:402
_invokeHandler @ rpcProtocol.ts:387
_receiveRequest @ rpcProtocol.ts:303
_receiveOneMessage @ rpcProtocol.ts:230
(anonymous) @ rpcProtocol.ts:105
fire @ event.ts:587
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
(anonymous) @ ipc.net.ts:592
fire @ event.ts:587
acceptChunk @ ipc.net.ts:239
(anonymous) @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:203
addChunk @ _stream_readable.js:295
readableAddChunk @ _stream_readable.js:276
Readable.push @ _stream_readable.js:210
onStreamRead @ internal/stream_base_commons.js:166
mainThreadExtensionService.ts:66 Error: Request error
	at Service.<anonymous> (/Users/erik/.vscode/extensions/ev3dev.ev3dev-browser-1.2.0/out/dnssd/dnssd-client.js:773)
	at Generator.next (<anonymous>)
	at fulfilled (/Users/erik/.vscode/extensions/ev3dev.ev3dev-browser-1.2.0/out/dnssd/dnssd-client.js:36)
	at processTicksAndRejections (internal/process/task_queues.js:85)

Any ideas?

 <https://user-images.githubusercontent.com/963645/88850739-63ae3800-d1b1-11ea-81db-58621e4a4726.png>

dlech, 2020-07-30:
What is the output if you run `dns-sd -B _sftp-ssh._tcp` in a terminal window?

uelundgren, 2020-07-30:
Last login: Thu Jul 30 20:33:44 on ttys000
erik@eriksmac ~ %  dns-sd -B _sftp-ssh._tcp
Browsing for _sftp-ssh._tcp
DNSService call failed -65547

dlech, 2020-07-30:
The fact that the command line didn't work either tells me there is a problem with your computer.

Google says `-65547` is `kDNSServiceErr_AlreadyRegistered` but I don't see any suggestions on how to fix it.

Have you tried rebooting your computer?

uelundgren, 2020-07-30:
Yes, tried rebooting several times - no difference.
Using Discovery, _sftp-ssh._tcp shows a whole bunch of addresses, see screenshot below.

dlech, 2020-07-30:
GitHub doesn't allow attachments to emails.

uelundgren, 2020-07-30:
Sorry. Reboot did not help.

The utility Discovery lists the following when browsing:

_sftp-ssh._tcp. (Secure File Transfer Protocol over SSH) - 1 item
EriksMac
EriksMac.local.
127.0.0.1:22
[::1]:22
[fe80::1]:22
192.168.1.124:22
[fe80::1cf4:c323:64e2:bd33]:22