ev3dev/ev3dev #1432: Creating a GUI on a PC that can interface with the EV3 Brick

bluemoon-123, 2020-08-06:
Hi there,

I'm currently working on my robot, which at the moment acts as a reactor system, using an SSH connection it can take in user input and deposit x mL of liquid into a reactor vessel. Is there a way that I can keep this user interface but use something like tkinter to create a GUI on my laptop which will be in direct contact with the EV3? Essentially just a fancy version of writing to a command line. 

Thanks guys.

dale1998, 2021-03-01:
Are you still interested in how to do this? Or did you manage to find a way, I am literally researching and messing around with tkinter at the moment

bluemoon-123, 2021-03-05:
Unfortunately my project finished a couple of months ago, and I never managed to figure out a solution (code still being run from the terminal). But if you ever find a way around this do let me know.

snaildos, 2021-03-25:
This is annoying. Make look at the code at the current GUI. We could somehow hijack the boot to run a custom script?