ev3dev/ev3dev #1433: UART sensors not communicating

JakubVanek, 2020-08-09:
**Describe the bug**
UART sensors are getting detected (port mode changes), but it seems that no actual communication takes place (`sensor<i>` does not get registered). EV3 touch sensor seems to work on the other hand.

**To Reproduce**
Steps to reproduce the behavior:
1. Download SD card image referenced below
2. Copy the image to the SD card using dd + enable 456 MHz overclock (but the bug occurs even without it) + copy `TIInit_6.2.31.bts` into /lib/firmware/ti-connectivity
3. Insert the SD card to the brick
4. Start the brick
5. Attach an EV3 sensor
6. See dmesg

The same thing happens after installing locally built ev3 kernel as well.

```
Please paste a simple program to reproduce the problem here!
```

**Expected behavior**
EV3 UART sensors work as expected.

**Screenshots**
dmesg output:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.85-ev3dev-3-beta2-ev3-dataage+ (kuba@oggy-kuba) (gcc version 8.3.0 (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36))) #1 PREEMPT Sun Aug 9 23:09:40 CEST 2020
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: LEGO MINDSTORMS EV3
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created DMA memory pool at 0x80000000, size 0 MiB
[    0.000000] OF: reserved mem: initialized node serial@80000000, compatible id shared-dma-pool
[    0.000000] DaVinci da850/omap-l138/am18x variant 0x1
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   DMA zone: 128 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 16384 pages, LIFO batch:3
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x414 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:   console=tty1 quiet
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 51020K/65536K available (5546K kernel code, 279K rwdata, 1816K rodata, 184K init, 155K bss, 14516K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
                   lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (5548 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 184 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 280 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 156 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] NR_IRQS: 325
[    0.000000] clocksource: timer0_1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000674] Console: colour dummy device 80x30
[    0.000726] console [tty1] enabled
[    0.000843] Calibrating delay loop... 226.09 BogoMIPS (lpj=1130496)
[    0.070121] pid_max: default: 32768 minimum: 301
[    0.070720] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070769] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.072984] CPU: Testing write buffer coherency: ok
[    0.076450] Setting up static identity map for 0xc0008400 - 0xc000843c
[    0.077048] rcu: Hierarchical SRCU implementation.
[... 205 lines cut ...]
 - Other software: SSH

**Robot (please complete the following information):**
 - Device: EV3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]
    - Linux kernel: stock `4.19.85-ev3dev-3-beta2-ev3`, locally-built `4.19.85-ev3dev-3-beta2-ev3-dataage+`
    - SD card image: https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2020-04-10/snapshot-ev3dev-buster-ev3-generic-2020-04-10.img.xz (after unxz-ing, I have md5sum of 045d8184187b1eec1f3a66040dc276bd)

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-buster-ev3-generic-2020-04-10
Kernel version:     4.19.85-ev3dev-3-beta2-ev3-dataage+
Brickman:           0.10.2
BogoMIPS:           226.09
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165344CF21
BOARD_INFO_TYPE=main
```

**Additional context**
I think the SD card may be the problem here, but I cannot prove or disprove that. I think I have used this firmware version successfully previously, but I'm not 100% sure. The brick and the sensors themselves work fine under stock firmware.

JakubVanek, 2020-08-10:
ev3dev-stretch on the same SD card appears to work - after connecting, the EV3 color sensor turns on its red LED after few seconds.
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2020-04-10
Kernel version:     4.14.117-ev3dev-2.3.5-ev3
Brickman:           0.10.3
BogoMIPS:           226.09
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165344CF21
BOARD_INFO_TYPE=main
```

dlech, 2020-08-10:
> [    0.143711] gpiochip_add_data_with_key: GPIOs 0..143 (davinci_gpio.0) failed to register, -517

`517` is `EPROBE_DEFER` and is normal. The other errors are probably the same.

dlech, 2020-08-10:
My first thought is that https://github.com/ev3dev/ev3-systemd/blob/ev3dev-stretch/systemd/ev3-uart%40.service is not working so the line discipline is not being attached.

JakubVanek, 2020-08-10:
It does look like that might be it, I will try to experiment with it tomorrow.

JakubVanek, 2020-08-11:
Indeed, `ev3-uart@.service` is the problem: 
```
robot@ev3dev:~$ sudo systemctl list-units --state=failed
  UNIT                             LOAD   ACTIVE SUB    DESCRIPTION                     
● ev3-uart@ev3x2dports:in1.service loaded failed failed LEGO MINDSTORMS EV3 UART Sensors

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

1 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```

```
robot@ev3dev:/lib/systemd/system$ sudo systemctl status ev3-uart@ev3x2dports:in1.service
● ev3-uart@ev3x2dports:in1.service - LEGO MINDSTORMS EV3 UART Sensors
   Loaded: loaded (/lib/systemd/system/ev3-uart@.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-08-11 07:07:38 UTC; 3min 41s ago
  Process: 575 ExecStart=/usr/sbin/ldattach 29 /dev/tty_ev3x2dports:in1 (code=exited, status=1/FAILURE)

Aug 11 07:07:38 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 UART Sensors...
Aug 11 07:07:38 ev3dev ldattach[575]: ldattach: cannot open /dev/tty_ev3x2dports:in1: No such file or directory
Aug 11 07:07:38 ev3dev systemd[1]: ev3-uart@ev3x2dports:in1.service: Control process exited, code=exited, status=1/FAILURE
Aug 11 07:07:38 ev3dev systemd[1]: ev3-uart@ev3x2dports:in1.service: Failed with result 'exit-code'.
Aug 11 07:07:38 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 UART Sensors.
```
The problem seems to be the `-` character encoding. The TTY does exists and running `sudo ldattach 29 /dev/tty_ev3-ports\:in1` successfully initializes the sensor. I tried changing between `%i` and `%I` in the unit file and it had no effect (with `systemctl daemon-reload` after edit).

Contents of `/run/udev/data/+lego\:ev3-ports\:in2\:ev3-uart-host` (I have reattached the sensor) are these:
```
robot@ev3dev:/run/udev/data$ cat +lego\:ev3-ports\:in2\:ev3-uart-host 
I:3253173330
E:SYSTEMD_WANTS=ev3-uart@ev3\x2dports:in2.service
E:SYSTEMD_ALIAS=/sys/subsystem/lego/devices/ev3_uart_host_ev3-ports:in2
G:systemd
```

At this point, I'm starting to feel the pain of people not liking systemd :smile: I tried searching for why the slash might be lost, but I couldn't find anything useful.

dlech, 2020-08-11:
Maybe caused by this change? https://github.com/systemd/systemd/commit/dcebc9bae4dcc3e844f01558c6127fc0d8745c8e

JakubVanek, 2020-08-15:
I don't think this change causes it, as this only affects the unit name if it is an empty template. If I use `SYSTEMD_WANT="ev3-uart@.service"`, I get `ev3-uart@sys-devices-platform-ev3\x2dports-ev3\x2dports:in3-lego\x2dport-port2-ev3\x2dports:in3:ev3\x2duart\x2dhost.service` as a full unit name. Unfortunately `SYSTEMD_ALIAS` is not considered for this, only the real sysfs path. At least we get the backslashes back, but the name would require some parsing. The mangling itself at the bottom of the patch doesn't seem to have an effect, as `/bin/systemd-escape --mangle 'ev3-uart@ev3\x2dports:in3.service'` returns the input string with slashes.

JakubVanek, 2020-08-15:
I think I found the issue:
The property content is unquoted [here](https://github.com/systemd/systemd/blob/61c8b73a35fd71cb7cbb41b78d93a124bf347e8e/src/core/device.c#L379). The unquoting itself happens [here](https://github.com/systemd/systemd/blob/f1cc283a5ab47aae64a2bdb4cd7f1916309906ff/src/basic/extract-word.c#L19). As only `EXTRACT_UNQUOTE` is specified, the backslash is discarded as no special unescaping mode is specified. Duplicating the backslash should work the problem around.

JakubVanek, 2020-08-15:
Workaround confirmed. By hardcoding this to `60-ev3.rules`, the port 4 got successfully activated:
```udev
SUBSYSTEMS=="lego", ENV{DEVTYPE}=="ev3-uart-host", ACTION=="add", TAG+="systemd", \
    ENV{SYSTEMD_WANTS}="ev3-uart@ev3\\x2dports:in4.service", \
    ENV{SYSTEMD_ALIAS}="/sys/subsystem/lego/devices/ev3_uart_host_$env{LEGO_ADDRESS}"
```

JakubVanek, 2020-08-15:
I have created https://github.com/systemd/systemd/issues/16735 to see if this is really the issue.

JakubVanek, 2020-08-15:
I have found a workaround that wouldn't require backporting the systemd fix:
`/lib/systemd/system/ev3-uart@.service`:
```
# This file is part of the ev3-systemd package

[Unit]
Description=LEGO MINDSTORMS EV3 UART Sensors
# This is the alias created by the ev3dev-uart udev rule.
BindsTo=%i.device
Wants=systemd-udev-settle.service
After=systemd-udev-settle.service

[Service]
Type=forking
ExecStart=/usr/share/ev3-systemd/ev3-start-uart.sh %I
```
part of `/lib/udev/rules.d/60-ev3.rules`:
```
# Start EV3 UART sensor line discipline when UART sensor is connected
SUBSYSTEMS=="lego", ENV{DEVTYPE}=="ev3-uart-host", ACTION=="add", TAG+="systemd", \
    ENV{SYSTEMD_WANTS}="ev3-uart@.service"
```
`/usr/share/ev3-systemd/ev3-start-uart.sh`
```
#!/bin/dash

# Line discipline helper for systemd
#
# Copyright (C) 2020 Jakub Vanek <linuxtardis@gmail.com>
#
# This program is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 2 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program; if not, see <http://www.gnu.org/licenses/>.

set -e

# command line parameters
sysfs_device="$1" # sysfs device path like sys/devices/platform/ev3-ports/ev3-ports:in3/lego-port/port2/ev3-ports:in3:ev3-uart-host
device_name="$(echo "$sysfs_device" | sed 's|:ev3-uart-host||g; s|sys/devices/platform/ev3-ports/ev3-ports:in./lego-port/port./||g')"
exec /usr/sbin/ldattach 29 "/dev/tty_$device_name"
```

I tried to add the second backslash in the udev rule, but the preceding backslashes would be always replaced by `_`.

JakubVanek, 2020-08-17:
The behaviour of systemd turned out to be correct. The `SYSTEMD_WANTS` value in udev can be enclosed in quotes and for this reason systemd eats the backslashes, marking the "escaped" characters as non-quotes.

On the other hand, there might another feature missing in udev and that is the ability to suspend invalid escape sequence replacement for `PROGRAM` output. This would allow us to pass `\\` using a sed script, but this seems not to be possible. `OPTIONS+="string_escape=none"` ought to enable this in v241 udev, but it doesn't work and v243+ removes the code that would have allowed this.

The proposed workaround may be the only viable solution then, as I don't think I want to backport new udev features to the stable debian version. It will splitting buster and stretch versions of the `ev3-systemd` package, as systemd in stretch (v232) does not contain https://github.com/systemd/systemd/commit/dcebc9bae4dcc3e844f01558c6127fc0d8745c8e.

JakubVanek, 2020-08-17:
I have prepared a fixed package at https://github.com/JakubVanek/ev3-systemd-private-dev/commits/ev3dev-buster . It fixes this bug by introducing the workaround and also it updates the packaging. Regarding copyright: I don't care about where it goes, I just felt that writing other's name as being responsible for the file and consequently all its errors is not the best thing. Feel free to unify it so that the package has only one copyright holder.

dlech, 2020-08-17:
I've actually been thinking about skipping ev3dev-buster and going straight to ev3dev-bullseye since bullseye will probably become stable in the next year or so. Would this help since bullseye has a newer version of systemd.?

JakubVanek, 2020-08-17:
I don't think so, current behaviour on buster corresponds to how systemd `master` would behave from looking into the source code. However, bullseye could still receive some new code to pass the double backslash from udev to systemd.

dlech, 2020-08-17:
I have published your changes to ev3dev-buster.

dlech, 2020-08-17:
doh, and I just accidentally deleted it

dlech, 2020-08-17:
and it is back again now

JakubVanek, 2020-08-17:
Thank you, I can confirm that running `sudo apt-get update && sudo apt-get dist-upgrade` on clean 2020-04-10 ev3dev-buster makes UART sensors work again.

closed 2020-08-17 by dlech

CarloWood, 2024-12-23:
Why was this closed? There is no way I can pass an escaped character with `SYSTEMD_WANTS`, and I tried *everything* in the past three hours.

My last attempt looks like this:
```
SUBSYSTEMS=="block", ENV{DEVTYPE}=="partition", ENV{ID_SERIAL_SHORT}=="408D5CBECBBDE7C0E9160666|408D5CBF5F0AE7C0E9150B70|408D5CBECAC3E7C0E9170AEC", PROGRAM="/usr/bin/systemd-escape 'foo-bar'", RUN+="/usr/local/sbin/usbtest '%c'", ENV{SYSTEMD_WANTS}="B@$result.service"
```
This results in `/usr/local/sbin/usbtest` being passed the correct value (`foo\x2dbar`) while the service `B@foox2dbar.service` is being started (removed backslash).

I tried adding backslashes, different quotes, using $result etc etc. Absolutely no way to get the right service started.

Doing `sudo systemctl start 'B@foo\x2dbar.service'` from the command line works as expected, so the bug is clearly with udev and SYSTEMD_WANTS.

Is there a solution for this?!

Using version 257