ev3dev/ev3dev #1434: EV3 Bluetooth firmware appears to be missing

JakubVanek, 2020-08-09:
**Describe the bug**
ev3dev-buster kernel prints a warning that it cannot load `TIInit_6.2.31.bts`. There is `TIInit_7.2.31.bts`, but kernel is not loading that one.

**To Reproduce**
Steps to reproduce the behavior:
1. Download SD card image referenced below
2. Copy the image to the SD card using dd + enable 456 MHz overclock
3. Insert the SD card to the brick
4. Start the brick
5. Run `dmesg`

```
Please paste a simple program to reproduce the problem here!
```

**Expected behavior**
Firmware is included in the SD card image. When I manually download the firmware from https://processors.wiki.ti.com/index.php/CC256x_Downloads , the problem disappears. The firmware is missing from the `firmware-ti-connectivity` ev3dev-buster package as well.

**Screenshots**
Warning from dmesg:
```
[   38.967606] Bluetooth: HCI UART driver ver 2.3
[   38.967649] Bluetooth: HCI UART protocol H4 registered
[   38.967667] Bluetooth: HCI UART protocol BCSP registered
[   39.037037] Bluetooth: HCI UART protocol LL registered
[   39.195805] hci-ti serial0-0: Direct firmware load for ti-connectivity/TIInit_6.2.31.bts failed with error -2
[   39.195962] Bluetooth: hci0: request_firmware failed(errno -2) for ti-connectivity/TIInit_6.2.31.bts
[   39.195990] Bluetooth: hci0: download firmware failed, retrying...
[   39.279639] hci-ti serial0-0: Direct firmware load for ti-connectivity/TIInit_6.2.31.bts failed with error -2
[   39.279705] Bluetooth: hci0: request_firmware failed(errno -2) for ti-connectivity/TIInit_6.2.31.bts
[   39.279729] Bluetooth: hci0: download firmware failed, retrying...
[   39.359646] hci-ti serial0-0: Direct firmware load for ti-connectivity/TIInit_6.2.31.bts failed with error -2
[   39.359711] Bluetooth: hci0: request_firmware failed(errno -2) for ti-connectivity/TIInit_6.2.31.bts
[   39.359732] Bluetooth: hci0: download firmware failed, retrying...
[   39.437830] hci-ti serial0-0: Direct firmware load for ti-connectivity/TIInit_6.2.31.bts failed with error -2
[   39.437894] Bluetooth: hci0: request_firmware failed(errno -2) for ti-connectivity/TIInit_6.2.31.bts
[   39.437918] Bluetooth: hci0: download firmware failed, retrying...
```

**Desktop (please complete the following information):**
 - OS: Xubuntu 18.04.5
 - Connection Type: USB ethernet adapter
 - Other software: SSH

**Robot (please complete the following information):**
 - Device: EV3
    - Linux kernel: stock `4.19.85-ev3dev-3-beta2-ev3`
    - SD card image: https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2020-04-10/snapshot-ev3dev-buster-ev3-

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-buster-ev3-generic-2020-04-10
Kernel version:     4.19.85-ev3dev-3-beta2-ev3
Brickman:           0.10.2
BogoMIPS:           226.09
Bluetooth:          
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165344CF21
BOARD_INFO_TYPE=main
```

**Additional context**
The firmware is missing from the image itself, not only from the SD card due to potential filesystem corruption.

JakubVanek, 2020-08-09:
Hmmmmmm, the `firmware-ev3` package is not installed and that package does contain the right firmware file.

JakubVanek, 2020-08-09:
https://github.com/ev3dev/docker-library/commit/283031fd7cce568aa2e173b5e751606cf13fde9a seems to be strongly related to this. I think the problem is that ev3dev repositories contain older version of the `firmware-ti-connectivity` package:
```
robot@ev3dev:~$ apt-cache policy firmware-ti-connectivity
firmware-ti-connectivity:
  Installed: 1:20161130-3+rpt3
  Candidate: 1:20161130-3+rpt3
  Version table:
 *** 1:20161130-3+rpt3 990
        990 http://archive.ev3dev.org/debian buster/main armel Packages
        100 /var/lib/dpkg/status
     20190114-2 500
        500 http://httpredir.debian.org/debian buster/non-free armel Packages
```

dlech, 2020-08-10:
Should be able to grab a newer package from buster-backports https://packages.debian.org/buster-backports/firmware-ti-connectivity

JakubVanek, 2020-08-17:
Actually, just installing `firmware-ti-connectivity` from buster (non-backports) fixes the problem too. I don't know if it is possible and if it would work, but withdrawing `firmware-ti-connectivity` from `ev3dev-buster` in ev3dev repos could be sufficient.

dlech, 2020-08-17:
> withdrawing `firmware-ti-connectivity` from `ev3dev-buster` in ev3dev repos could be sufficient.

Done. Please reopen if that didn't fix it.

closed 2020-08-17 by dlech

JakubVanek, 2020-08-17:
I am sorry, I have only now realized that the `1:...` in the `apt-cache policy` is likely the epoch number (of which I have only read recently) of the ev3dev version and not some status reported by the command. Therefore I think that the currently locally installed version 1:2016... will take precedence over the "new" 0:2019... upstream one. I originally thought that the precedence is from the `ev3dev` origin having higher priority than "debian" origin, but this may not be the case.

JakubVanek, 2020-08-17:
Currently new `ev3dev-buster` snapshots will be fixed (as they have no way of installing the `1:2016...` package), but existing snapshots are stuck at the `1:2016...` version.