ev3dev/ev3dev #1435: Install pybricks-micropython on top of ev3dev2

ivan-marroquin, 2020-08-10:
**Question**
Hi all,

I am using ev3dev2 version 4.14.117-ev3dev-2.3.4-ev3, and I would like to know if there is a way to add 
pybricks-micropython firmware.

Many thanks,

Ivan

**Background Information**

I am trying to ... because I want to build a robot that ...

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
 - Software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
 - Versions: [see below - SD card image version, Linux kernel version, etc.]

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

dlech, 2020-08-10:
You can ensure that the most recent version of `pybricks-micropython` is installed by making sure your EV3 is connected to the internet, then run `sudo apt update && sudo apt install pybricks-micropytyhon` in an SSH terminal.

ivan-marroquin, 2020-08-12:
Hi @dlech 

Many thanks for the advice!

Ivan

closed 2020-08-12 by ivan-marroquin