ev3dev/ev3dev #1436: Light Sensor Array Driver

svigo, 2020-08-16:
I want to use the LSA that i successfully use in NXT

http://www.mindsensors.com/ev3-and-nxt/47-light-sensor-array-for-nxt-or-ev3

I'm a computer engineering student and have some experience programming in C perhaps with some help i can make a driver for this sensor to ev3dev
Where i have to begin?

JakubVanek, 2020-08-17:
Hi @svigo,

I don't have much experience with ev3dev drivers either, but [I2C Devices](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/i2c.html), [nxt_analog_sensor_defs.c](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-buster/sensors/nxt_analog_sensor_defs.c) and [nxt_i2c_sensor.h](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-buster/sensors/nxt_i2c_sensor.h) should be good starting points.  Many of the sensors use a declarative approach to their drivers as their principle of operation is quite similar. The NXT I2C core then handles the mode switching etc. based on the data specified in `nxt_analog_sensor_defs.c`. On the hardware side, the NXC driver (mostly `LSA-lib.nxc`) should provide the necessary information regarding the I2C registers and commands.

Hope it helps,

Jakub

dlech, 2020-08-17:
There is already a driver for the Mindsensor Light Sensor Array.

http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ms-light-array

svigo, 2020-08-18:
Oh, excelent!