ev3dev/ev3dev #1442: Realtek RTL8188CUS based Wi-Fi won't work (Edimax EW-7811Un).

ykhokhlov, 2020-10-16:
Hi!

Summary
==============================
I've bought a Wi-Fi dongle. Edimax EW-7811Un.

Flashed ev3dev-stretch-ev3-generic-2020-04-10.zip image.
(4.14.117-ev3dev-2.3.5-ev3)

I've tried UI, but have got "Not Available" in the Wi-Fi menu.

Then I've gone to command-line.
All the details you will find below.
Even after `modprobe 8192cu` and `usbcore: registered new interface driver rtl8192cu` things a still without Wi-Fi.

Why do you say that Edimax EW-7811Un 802.11n Wireless Adapter is supported?

Technical details
==============================

```
robot@ev3dev:~$ uname -r
4.14.117-ev3dev-2.3.5-ev3
```

Have tried connmanctl with no luck.

```
robot@ev3dev:~$ connmanctl
Error getting VPN connections: The name net.connman.vpn was not provided by any .service filesconnmanctl> enable wifi
Error wifi: Method "SetProperty" with signature "sv" on interface "net.connman.Technology" doesn't exist
```

```
robot@ev3dev:~$ lsusb
Bus 001 Device 003: ID 7392:b811 Edimax Technology Co., Ltd
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
...

```
robot@ev3dev:~$ dpkg --list | grep 8192
robot@ev3dev:~$ sudo dpkg --list | grep 8192
```

...

`robot@ev3dev:~/rtl8192cu$ sudo modprobe 8192cu`
...

```
robot@ev3dev:~/rtl8192cu$ lsmod
Module                  Size  Used by
8192cu                552861  0
cfg80211              255501  1 8192cu
iptable_nat             1728  0
nf_conntrack_ipv4       5927  1
nf_defrag_ipv4          1634  1 nf_conntrack_ipv4
nf_nat_ipv4             5140  1 iptable_nat
nf_nat                 14660  1 nf_nat_ipv4
nf_conntrack           87726  3 nf_conntrack_ipv4,nf_nat_ipv4,nf_nat
libcrc32c               1100  2 nf_conntrack,nf_nat
usb_f_rndis            15528  2
usb_f_ecm               6278  2
u_ether                12306  2 usb_f_ecm,usb_f_rndis
ev3_uart_sensor_ld      7503  2
ev3_uart_sensor        13506  1 ev3_uart_sensor_ld
nxt_i2c_sensor         35789  1 ev3_uart_sensor
servo_motor_class       6061  1 nxt_i2c_sensor
uinput                  8046  0
[... 302 lines cut ...]
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 68  bytes 8771 (8.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 68  bytes 8771 (8.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=-28669<UP,BROADCAST,MULTICAST,DYNAMIC>  mtu 1500
        ether 02:16:53:5e:02:bb  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb1: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet 192.168.137.101  netmask 255.255.255.0  broadcast 192.168.137.255
        inet6 fe80::2016:53ff:fe5e:2bb  prefixlen 64  scopeid 0x20<link>
        ether 22:16:53:5e:02:bb  txqueuelen 1000  (Ethernet)
        RX packets 14856  bytes 12632709 (12.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 7531  bytes 1391245 (1.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

dlech, 2020-10-16:
> Bus 001 Device 003: ID 7392:b811 Edimax Technology Co., Ltd

Sorry, but this ID (7392:b811) is not listed in the [list of supported and known working dongles](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles).

ykhokhlov, 2020-10-16:
Oh!
Is it possible to add such support? What do you recommend to do?
Yes, I can sell this dongle and buy another :-) (good alternative solution)

ykhokhlov, 2020-10-16:
Oh!
Is it possible to add such support? What do you recommend to do? @dlech 
Yes, I can sell this dongle and buy another :-) (good alternative solution)

dlech, 2020-10-16:
> Yes, I can sell this dongle and buy another :-) (good alternative solution)

That is probably the best option. Web search doesn't come up with much for 7392:b811.

xdsl, 2020-12-16:
> > Bus 001 Device 003: ID 7392:b811 Edimax Technology Co., Ltd
> 
> Sorry, but this ID (7392:b811) is not listed in the [list of supported and known working dongles](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles).

ev3dev-jessie not support, but ev3dev-stretch support:
0bda:0179 Realtek Semiconductor Corp. RTL8188ETV Wireless LAN 802.11n Network Adapter
148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter

xdsl, 2020-12-16:
0bda:0179 and 148f:7601 added to [wiki](https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles)

ykhokhlov, 2021-01-01:
> ev3dev-jessie not support, but ev3dev-stretch support:
> 0bda:0179 Realtek Semiconductor Corp. RTL8188ETV Wireless LAN 802.11n Network Adapter
> 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
> 0bda:0179 and 148f:7601 added to wiki
But I have 7392:b811. How this helps me?

xdsl, 2021-01-02:
may be help:
```
modprobe 8188eu
echo 7392 b811 /sys/bus/usb/drivers/r8188eu/new_id
```
see:
https://www.mail-archive.com/bugs@openbsd.org/msg15711.html
https://lwn.net/Articles/160944/

xdsl, 2021-01-10:
I compile rtl8188eu driver from https://github.com/lwfinger/rtl8188eu for latest ev3dev-stretch kernel 4.14.117-ev3dev-2.3.7-ev3 (https://github.com/ev3dev/ev3dev/issues/1458#issuecomment-753652773)
It support Edimax EW-7811Un V2: http://rubirobot.ru/files/kernel/modules/4.14.117-ev3dev-2.3.5-ev3/rtl8188eu/support.txt
Download: http://rubirobot.ru/files/kernel/modules/4.14.117-ev3dev-2.3.5-ev3/rtl8188eu/8188eu.ko
Replace current driver: /lib/modules/4.14.117-ev3dev-2.3.5-ev3/drivers/net/wireless/8188eu.ko
Reboot and Pray)

ykhokhlov, 2021-01-21:
@xdsl, thank you for your help!
Now it works like a charm! Nice!

Here what I've done. This may help those, who have encountered the same problem.

1. Downloaded 8188eu.ko (provided by @xdsl): http://rubirobot.ru/files/kernel/modules/4.14.117-ev3dev-2.3.5-ev3/rtl8188eu/8188eu.ko (I've attached it here for some case).
[8188eu_drv.zip](https://github.com/ev3dev/ev3dev/files/5847952/8188eu_drv.zip)
2. Replaced the current driver in this folder (I've backed up the old one): /lib/modules/4.14.117-ev3dev-2.3.5-ev3/drivers/net/wireless/8188eu.ko  
3. Heve done a reboot of the EV3.
4. WiFi option is still empty.
5. After eject and inserting back the USB Dongle:
```
[  462.186457] usb 1-1: USB disconnect, device number 2
[  465.174939] usb 1-1: new full-speed USB device number 3 using ohci-da8xx
[  465.448181] usb 1-1: New USB device found, idVendor=7392, idProduct=b811
[  465.448293] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  465.448335] usb 1-1: Product: Edimax N150 Adapter
[  465.448403] usb 1-1: Manufacturer: Realtek
[  465.448442] usb 1-1: SerialNumber: 08BEAC114ECB
```
6. Executed this command: `robot@ev3dev:~$ sudo modprobe 8188eu`
```
[  572.110892] 8188eu: loading out-of-tree module taints kernel.
[  572.185259] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[  572.497685] EEPROM ID = 0x8129
[  572.539805] usbcore: registered new interface driver r8188eu
[  573.206454] r8188eu 1-1:1.0 wlx08beac114ecb: renamed from wlan0
```
7. **After this WiFi becomes available in UI.**
These lines were added to `dmesg` output after powering on the WiFi Dongle via UI:
```
[  696.028998] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[  697.874351] MAC Address = 08:be:ac:11:4e:cb
[  697.949263] IPv6: ADDRCONF(NETDEV_UP): wlx08beac114ecb: link is not ready
[  701.008037] R8188EU: INFO indicate disassoc
[  812.424762] R8188EU: INFO assoc success
[  812.661659] IPv6: ADDRCONF(NETDEV_CHANGE): wlx08beac114ecb: link becomes ready
```
8. I've used UI to power up WiFi, scan for networks, and connector to one of them.
[image]

@xdsl, may want to add something like advice and after this, I plan to close this issue.

xdsl, 2021-01-21:
You can also use my custom built ev3dev kernel: http://rubirobot.ru/files/kernel/4.14.117-ev3dev-2.3.5-rrsport-1.0-ev3+/
Download http://rubirobot.ru/files/kernel/4.14.117-ev3dev-2.3.5-rrsport-1.0-ev3+/linux-image-4.14.117-ev3dev-2.3.5-rrsport-1.0-ev3+_12_armel.deb and install it
Main feature: ipv6 in module and may be fully disabled - https://github.com/ev3dev/ev3dev/issues/1458#issuecomment-753520103

ykhokhlov, 2021-01-21:
@xdsl, Thank you! I'll try it soon.
P.S.: http://rubirobot.ru/ - looks very useful for me. Thank you for this reference and your contribution!

closed 2021-01-21 by ykhokhlov

ykhokhlov, 2021-01-21:
FYI: @dlech

G1den, 2024-12-10:
> > @xdsl, thank you for your help! Now it works like a charm! Nice!
> > Here what I've done. This may help those, who have encountered the same problem.
> > 
> > 1. Downloaded 8188eu.ko (provided by @xdsl): http://rubirobot.ru/files/kernel/modules/4.14.117-ev3dev-2.3.5-ev3/rtl8188eu/8188eu.ko (I've attached it here for some case).
> >    [8188eu_drv.zip](https://github.com/ev3dev/ev3dev/files/5847952/8188eu_drv.zip)
> > 2. Replaced the current driver in this folder (I've backed up the old one): /lib/modules/4.14.117-ev3dev-2.3.5-ev3/drivers/net/wireless/8188eu.ko
> > 3. Heve done a reboot of the EV3.
> > 4. WiFi option is still empty.
> > 5. After eject and inserting back the USB Dongle:
> > 
> > ```
> > [  462.186457] usb 1-1: USB disconnect, device number 2
> > [  465.174939] usb 1-1: new full-speed USB device number 3 using ohci-da8xx
> > [  465.448181] usb 1-1: New USB device found, idVendor=7392, idProduct=b811
> > [  465.448293] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> > [  465.448335] usb 1-1: Product: Edimax N150 Adapter
> > [  465.448403] usb 1-1: Manufacturer: Realtek
> > [  465.448442] usb 1-1: SerialNumber: 08BEAC114ECB
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 6. Executed this command: `robot@ev3dev:~$ sudo modprobe 8188eu`
> > 
> > ```
> > [  572.110892] 8188eu: loading out-of-tree module taints kernel.
> > [  572.185259] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
> > [  572.497685] EEPROM ID = 0x8129
> > [  572.539805] usbcore: registered new interface driver r8188eu
> > [  573.206454] r8188eu 1-1:1.0 wlx08beac114ecb: renamed from wlan0
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 7. Added a line to `/etc/modules` with `8188eu`. This enables autoload of this module on ev3dev start.
> > 8. **After this WiFi becomes available in UI.**
> >    These lines were added to `dmesg` output after powering on the WiFi Dongle via UI:
> > 
> > ```
> > [  696.028998] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
> > [  697.874351] MAC Address = 08:be:ac:11:4e:cb
> > [  697.949263] IPv6: ADDRCONF(NETDEV_UP): wlx08beac114ecb: link is not ready
> > [  701.008037] R8188EU: INFO indicate disassoc
> > [  812.424762] R8188EU: INFO assoc success
> > [  812.661659] IPv6: ADDRCONF(NETDEV_CHANGE): wlx08beac114ecb: link becomes ready
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 9. I've used UI to power up WiFi, scan for networks, and connector to one of them.
> >    [image]
> > 
> > @xdsl, may want to add something like advice and after this, I plan to close this issue.
> 
> Hi,
> 
> Could someone explain, how to replace the driver, if I have no permission for that? The old root's password r***e doesn't work.
> 
> Thanks in adavance, Vroobel

The root password on a flashed ev3dev is "maker". Us an sftp client like psftp to transfer the driver to the folder. I put the driver in a folder that i can access without root and used "sudo cp" to transfer the file to the folder.

lchaubert, 2025-02-27:
Thank you so much, working perfectly also with an ASUS USB-N10 Nano rev B1 (0b05:18f0) (Based on REALTEK RTL8188EUS)