ev3dev/ev3dev #1443: bluetooth connection bus isn't avalible

ev3dever, 2020-10-20:
**Describe the bug**
A clear and concise description of what the bug is.
when I go to the bluetooth connection on my ev3 brick with ev3dev sd card in it it says when I try to connect with bluetooth bluetooth protocol not avalible

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'bluetooth on the ev3 brick'
2. Click on 'your device'
3. Scroll down to 'connect'
4. See error

```
Please paste a simple program to reproduce the problem here!
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Window 10, macOS 10.14]
windows 64 bit
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi
bluetooth
 - Other software: [e.g. Visual Studio Code, PuTTY]

**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
ev3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]
most recent one ( i got it 2 days ago)

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

**Additional context**
Add any other context about the problem here.

dlech, 2020-10-20:
What do you want to do with Bluetooth? Have you followed these instructions? https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/

The *Connect* button is usually only used for connecting the EV3 to devices like keyboards and game controllers.

ev3dever, 2020-10-20:
okay I tried to set up connect automatically but when I click the network connection button nothing happens

ev3dever, 2020-10-20:
what I am trying to connect to with Bluetooth is my computer

snaildos, 2021-03-25:
Same. "Timeout was reached".