ev3dev/ev3dev #1444: ev3dev file browser

ev3dever, 2020-10-20:
**Question**

How do you go back from a file?

**Background Information**

I am trying to run programs because I want to build a robot that connects with Alexa but, I go in a select mission01.py but I cant go back out to the previous folder I have to reboot ev3 dev to go to the previous folder and it is so annoyingg have to restart it every time I want to run a program.

**Screenshots**
If applicable, add screenshots to help explain your question.

**Desktop (please complete the following information):**
 - OS and version: [e.g. Window 10 1809, macOS 10.14]
ev3 brick with ev3dev
 - Connection Type: [e.g. USB, Bluetooth, Wi-Fi]
ev3brick with ev3dev

 - Software: [e.g. Visual Studio Code, PuTTY]
ev3dev on the brick
**Robot (please complete the following information):**
 - Device: [e.g. EV3, BrickPi3]
ev3
 - Versions: [see below - SD card image version, Linux kernel version, etc.]
most recent got it 2 days ago

```
Please paste the system info from the ev3dev VS Code extension or the output of `ev3dev-sysinfo -m` here!
```

dlech, 2020-10-20:
Select the `../` folder in the *File Browser* to go up one directory level.

ev3dever, 2020-10-20:
oh, that's what that button is for thanks!

closed 2020-10-20 by ev3dever